emericg/WatchFlower #1: Logo Proposal

Tobaloidee, 2019-04-19:
Hello @emericg  Good day!! I am a graphic artist and an open source enthusiast.. I would like to  contribute to your project by designing a logo for free. Actually I already designed one for this project as I found it interesting. I hope you like it and consider using it! -[Tobaloidee](https://steemit.com/@tobaloidee)

- The design is simple but memorable. It is unique as it embodies the initial of your project which is "W" placed in part of the flower. Still open for your suggestions if you have any. :)

[image]

If you like it, I will finalize the files and make a PR for it!

Tobaloidee, 2019-04-19:
I understand @emericg  I will check your other project and try to make a logo for it! closing this issue now.

closed 2019-04-19 by Tobaloidee

emericg, 2019-04-27:
Yes it's a really cool icon, and thank you for taking some time to work on this.
I'm sorry but as we discussed, a whole new visual design for both icon and UI are already well underway, and so I cannot accept your proposition :(