emericg/WatchFlower #2: [Feature request] Live mode

gonewild22, 2019-08-08:
What a great android app! I was looking for such an open source solution for a very long time, so thank you so much! :) One function I'm really missing: I'm doing a lot of EC measurements for my hydro plants. Therefore I have to check my EC values regulary. Is it possible to integrate a kind of "live mode" which shows all values on one screen and updates them every 2 seconds or so?! (like in Xiaomi's official app) That would be really awesome!

emericg, 2019-08-09:
Hi! I'm glad you like the app.
So regarding a live mode, there is two ways this could be accomplished. 
- The first one is just setting the update timers from 30-180min to 10s. If you build the app yourself, it should be easy to do (as a short term solution). However, this would be a significant drain on both phone and sensors batteries...
- The second (and prefered) method would be the ability to read Bluetooth low energy "advertisements" messages. The sensors are broadcasting such messages regularly, without the need to pair the phone and the sensor to do a reading. I've started working on this already, but the problem with this is that ble advertisements support has just been added to Qt 5.12 and in my (limited) testing, it just doesn't work :(

So I'll continue to see if I can get the ble advertisements working, because a live mode like you describe would be pretty great (especially for thermometers), but the worst case scenario would be to have to wait for better Qt support.

gonewild22, 2019-09-26:
It would be great if you could at least push the refresh button more than one time per minute manually. Otherwise the status switches to "Just synced!" and you have to wait before you can measure / sync again. Would that be possible?
https://i.postimg.cc/kMWyJPXK/Watch-Flower-Screen.png

emericg, 2019-09-28:
I've made the change but I'm not really happy about one thing :
The idea is that if you click on "refresh all" and then some of the sensor fails to refresh, you could just click again and only the failed device(s) would try again to refresh. With the current modification, now all of the devices are refreshed again, even if they where just successfully updated.

I may allow more of one update per minute only when the refreshed are triggered directly from the device's screen, not on the main screen with the "refresh all" button. Would the be OK for you?

gonewild22, 2019-09-30:
This sounds more than great! =) Thanks for asking & adopting! I really appreciate your help & support!

emericg, 2019-10-07:
No problem, I'm glad you like the software ^^
If you are using the play store version, let me know if you need me to make a new release.

Regarding a "proper" live mode I didn't made much progress so I'll let this ticket open until then.

gonewild22, 2019-10-11:
I'm using the play store version, jep. But no stress, I'm very grateful already. So whenever you have time! :) Thx again!

emericg, 2019-10-15:
I finished another round of fixes and improvements, I'll push a new version very soon!

emericg, 2021-06-15:
The live mode is finally here (well, in the 2.0 release, available very very soon). Time to close our longest running issue ^^

closed 2021-06-15 by emericg