emericg/WatchFlower #3: Feature Request - update other tools

eserero, 2019-09-27:
hi - WatchFlower works great on my windows machine. I"m also running a home automation software (Domoticz), and it will be great to integrate watchflower to it. any way I can configure a notification address (say REST message) or be able to pull the data collected?
 
This will allow me to integrate this to other automation flows (like sending mails when a plan need water etc.

emericg, 2019-09-28:
Hi!
Wow I actually had no idea that it worked for real on windows :)

I know there is a BLEA plugin that work with the xiaomi bluetooth sensors for Jeedom, I guess there should be a more "direct" support for domoticz too?

Regarding adding a bridge between WatchFlower and Domotiz (or other things) it should be possible but that really wasn't a goal of this application.
If it is just sending a REST message, it should be pretty straightforward to do however, but what would be the format/content of that message?
Pulling or sharing the datas should also be possible. The app write them in a sqlite database wich can be opened by other applications. Also, using a real sql database and even sharing it between instances of WatchFlower or other software is completely possible, but again WatchFlower wasn't supposed to become a cloud platform ^^

eserero, 2019-10-04:
Hi - thanks for the answer - and sorry I took my time answering.
well - 2 things happened after I sent you my request. the first is that I found your database, and was able to create a small node service that just read it. now its pretty simple to poll the database every once in a while and update domoticz. so no real need for an API (at least for me). The other thing is that it stopped working - my mac will just not connect anymore to the xiaomi plan thingy (my phone have no problem...). I think I had it with domoticz on windows...  if I move to linux, I can just use any some python plugin in domoticz...
thanks for your help anyway. and great little program :-)

here is the query i use - just for reference... 
select * from devices, datas where devices.deviceAddr=datas.deviceAddr

emericg, 2019-10-15:
Allright so I'm a little lost, do you want to try debug the macOS version or can we close this issue?
Glad you like the application ^^

closed 2019-11-13 by emericg