emericg/WatchFlower #4: Windows app find flower care module but no data

bvalat, 2020-02-25:
The last version detect all my flower care module but fail to retreive data.
The module is connecting, updating and then go offline without any change...
[image]
Is there any log or debug mode to find what's happening?

emericg, 2020-03-07:
Hi,
Did it used to work with a previous version ? I never test for regressions on windows because WatchFlower bluetooth never worked on mine, but I have at least one report saying it was working with windows. Also I'm currently using an Asus BT400 too, so there's a chance...
There are logs outputted when errors happens, especially on the BLE parts, but again I have no idea if they even show on widnows. Are you able to start the app in a console or something?

I will have a look at this as soon as I'm done with my current freelance job.

bvalat, 2020-03-07:
No, it was my first time use.
It's not working on all my computers on win 10 64biis FR and i tried all realease versions.
I have an error loop in debug.log : 
[0225/213843.734:ERROR:crash_report_database_win.cc(469)] failed to stat report

By the way I managed to get it working with a live ubuntu 19.10 so I think bluetooth part is OK.

emericg, 2020-03-08:
Allright thanks for your feedback, I'll have a look at the windows version by the end of this week.

emericg, 2020-03-19:
By the way I had a look at this the other day, I can confirm that I have the exact same problem as you.
However I have no idea why it's not working. Something in the windows bluetooth backend doesn't behave like in all of the other platforms, so it might take more than a quick fix...

bvalat, 2020-03-25:
OK thanks for the update.
Can you tell me how you compile the project on windows?
Maybe i'll find some time to do some debug.

emericg, 2020-03-25:
Do you have Qt installed ? if yes, you juste load the .pro file and run, there is no external dependancies.
Otherwise, what I did last week, uninstalled everything that I had on an old windows install, then installed MSVC community 2019, Qt 5.14 (with Qt charts) msvc2017 integration (works with 2019 apparently).
If you don't have or don't want msvc, you can install mingw64 from the Qt installer too, I see no reason why that shouldn't work.
Then just load the .pro in Qt creator and everyting worked without manual steps (I was surprised).

From what I can gather, the bluetooth adapter handling is OK, the scanning/connection to the devices works, then service discovery is ok too, but when we ask for details about the services (last step before reading datas) there it just stop there. The discovery starts but never send an OK signal, which should not happen obviously...

bvalat, 2020-03-25:
I never used QT, last time i made some dev was java 10 years ago...
But i was curious to compile it myself and dig into the code.
Maybe with some tests i'll find something (i have good instinct).
And your project look promissing for my need. 
I just want to have a program polling my sensors and allert me when water is needed, i'll try to modify your app. If i manage to make something not too ugly, will push you my code.

emericg, 2020-03-25:
That's cool, you have homemade sensors?
There is an abstraction for devices (device.h) that makes it easy to add many BLE devices with different capabilities. Even wifi devices (or anything really) should not be hard to implement. If you need help with anything let me know.

For this particular problem, there is an example program, "BLE Scanner Example" (you can access it within Qt Creator example panel) that I used as a base for WatchFlower 2 years ago, and it seem to work under windows, so analysing the differences between WatchFlower and this example should provide a solution.

bvalat, 2020-03-25:
No i have hhcc mi flora sensors.
But never found an easy way/app to send me notification to give water to my plants.
Flower Care and Mi Home apps are useless in this case.

Let's focus on my first problem : windows app not working. :-)

kebasaa, 2020-04-06:
I confirm this bug, it really doesn't work on Windows 10

unicumnext, 2020-04-25:
Hy! I confirm too, it is not work.

bvalat, 2020-06-24:
I have same problem with LYWSD03MMC .
The device is detected. It connects to it (i see the bluetooth icon on the device) but no data are fetched and device is shown offline after that...
So it seems to be a global (QT?)  bluetooth issue on windows.

bvalat, 2020-06-24:
I manage to get LYWSD03MMC  by starting watchflower with admin rights.
Doesn't seem to work each time but i have now 2 thermometer synchronised. 
No luck for flower care.

EchoFPV, 2020-07-22:
Confirm the same issue/error on v0.14

PowerCoder, 2020-08-01:
So Windows 10 has issues? Because mine has never been able to even detect any of my 3 VegTrugs. I already switched to a BT5 dongle, verified BLE, tweaked the Privacy settings to see if maybe the app needed a permission. He just won't discover them.

EchoFPV, 2020-08-05:
> So Windows 10 has issues? Because mine has never been able to even detect any of my 3 VegTrugs (HHCCJCY01HHCC). I already switched to a BT5 dongle, verified BLE, tweaked the Privacy settings to see if maybe the app needed a permission. He just won't discover them.

I'm on Windows 10, detects sensor but same issue as original post :(

emericg, 2020-08-05:
@PowerCoder Your dongle is probably just not compatible. Can you tell us what model it is?

I'll try to have another look at this problem today.

PowerCoder, 2020-08-05:
Just your basic USB Generic Bluetooth Radio from the computer store. If you told me what specific information you are looking for, I'll see if I can find it. Other than that, there's not much to see. Plus, I tried it on a BT4 dongle as well. Neither work. Works fine with my other devices though. I'll try my Windows 10 tablet as well.

Thanks for the effort! It would be great if I could watch directly from my computer.

bvalat, 2020-08-05:
I have the issue with asus bt400, laptop with intel bt, mini pc, ....
All my bt windows computers have trouble.
If i use a linux virtual machine on the same pc, it works.

bvalat, 2020-08-05:
Today i downloaded v0.14 let it detect my flowercare and LYWSD03MMC 
And then i removed all flowercare and kept only the LYWSD03MMC.
To my surprise i works with my Asus bt400
[image]

PowerCoder, 2020-08-05:
Looks like I missed an update. On 0.14 I have the same problem. My VegTrugs are detected right away. They switch from connecting to synchronizing to offline. The one already programmed on my phone does nothing, the new one I just got does three blinks pause three blinks (probably because it is not programmed yet).

FYI, It's a Windows 10 1909, Flower Care 0.14 running as admin.

emericg, 2020-08-07:
OK it look like it was (still is) some sort of Qt bug (https://bugreports.qt.io/browse/QTBUG-80770) that needed a workaround.
That should be fixed, you can grab a release from the CI service here and test it https://ci.appveyor.com/project/emericg/watchflower/build/artifacts

emericg, 2020-08-07:
> Today i downloaded v0.14 let it detect my flowercare and LYWSD03MMC
> And then i removed all flowercare and kept only the LYWSD03MMC.
> To my surprise i works with my Asus bt400

I tried the same thing (same adapter and same sensor) and it still wasn't working for me, but that lead me to a timing related problem... Thanks for investigating ^^

PowerCoder, 2020-08-07:
Still the same problem with 0.15-447. Do I need to remove them first? If so, where is that configuration stored?

EchoFPV, 2020-08-08:
> Still the same problem with 0.15-447. Do I need to remove them first? If so, where is that configuration stored?

Same issue, where are the settings stored???

EchoFPV, 2020-08-08:
> > Still the same problem with 0.15-447. Do I need to remove them first? If so, where is that configuration stored?
> 
> Same issue, where are the settings stored???

SOLVED: delete Watchflower directories from C:\Users\[your user name]\AppData\Local and from the roaming directory. Relaunch and all works.

emericg, 2020-08-08:
Well it shouldn't have anything to do with settings, but as long as it works ^^
Maybe some kind of bluetooth related cache? I've had this problem before on linux, very annoying to debug...

PowerCoder, 2020-08-08:
Good news, it still doesn't work for me! :)

Queued -> Connecting -> Offline. I guess that means it isn't settings after all. Although I'm inclined to create a feature request to make the app fully portable, with settings stored with the app.

EchoFPV, 2020-08-08:
> Good news, it still doesn't work for me! :)
> 
> Queued -> Connecting -> Offline. I guess that means it isn't settings after all. Although I'm inclined to create a feature request to make the app fully portable, with settings stored with the app.

Remove all folders under appdata relating to Watchflower and then download and run v15. mine is working now :)

[image]

EchoFPV, 2020-08-09:
stopped working, 2 hours ago.. sync and then nothing...

emericg, 2020-08-09:
I've tried to add another workaround (applied just for flower care devices ATM), can you test the latest version?
Also I really don't think it has anything to do with the AppData folder, just timing.

EchoFPV, 2020-08-09:
> I've tried to add another workaround (applied just for flower care devices ATM), can you test the latest version?
> Also I really don't think it has anything to do with the AppData folder, just timing.

I'm using the v15, unless there is one newer?

emericg, 2020-08-09:
I'm referring to the very latest build, 0.15-450
https://ci.appveyor.com/project/emericg/watchflower/build/artifacts

v15 is just the version number that will be used to identify the next release

EchoFPV, 2020-08-09:
> I'm referring to the very latest build, 0.15-450
> https://ci.appveyor.com/project/emericg/watchflower/build/artifacts
> 
> v15 is just the version number that will be used to identify the next release

OK, will give it a test

PowerCoder, 2020-08-09:
Still the same problem. One thing changed though. Previously it would only detect the sensor if I held it right next to the BT receiver. Now it detects the sensor already in use in a plant nearby.

unicumnext, 2020-08-12:
> I'm referring to the very latest build, 0.15-450
> https://ci.appveyor.com/project/emericg/watchflower/build/artifacts
> 
> v15 is just the version number that will be used to identify the next release

A ty it and works for me.
First time I removed the C:\Users[your user name]\AppData\Local and from the roaming directory.
Then I run it.

emericg, 2020-08-12:
Thanks for testing. I **think** the issue is fixed as well as it can be, the remaining failures are juste out of range or low signal problems. Probably.

The content of the AppData\Local hasn't much to do with this problem. Removing the settings and database is completely unrelated. There should be a cache\ directory in there, but even removing that should not impact anything else than the UI performances.
Just launching a new scan (not just an update) should be enough. Worst case you can remove the sensors in the app and launch a new scan (and that doesn't delete the history).

unicumnext, 2020-08-13:
Ok it's working, but many times crashed, then exit when update the data.

PowerCoder, 2020-08-18:
I used it on a Windows 10 tablet. After failing to connect twice it finally synced. Once, after that it crashes every single time it tries to sync.

nerudesu, 2020-09-04:
> I'm referring to the very latest build, 0.15-450
> https://ci.appveyor.com/project/emericg/watchflower/build/artifacts
> 
> v15 is just the version number that will be used to identify the next release

It's confirmed! Using this build and it successfully synced. Don't forget to delete previous data on the AppData folder first, both for local and roaming.

emericg, 2020-09-04:
Hi, thanks for testing!
Crashes are a surprise because I usually don't have any, I'll try to test with other windows computers to see if I can find something.

emericg, 2020-12-07:
There have been no complaints for a while about this, I'll close this issue.
Feel free to reopen if new Windows bugs are found.

closed 2020-12-07 by emericg