emericg/WatchFlower #5: Plant database en share info online

svole1982, 2020-03-30:
Hello,

I downloaded your app to used is for my VegTrug plant sensor.

Can you also connect a specific plant to the sensor from a plant database. At the moment it’s only a plain text field to fill in a regular plant name?

Can you also share the info from te app on a website or social media, live overview?

Thx in advance

emericg, 2020-03-31:
Hi, does the VegTrug plant sensor work well with the application? I never tried with one.

> Can you also connect a specific plant to the sensor from a plant database. At the moment it’s only a plain text field to fill in a regular plant name?

Yes at the moment you must set your plant name and limits manually.
Unfortunately I don't know of any open plant database, and doing this work manually is a huge undertaking. But that is something that's always in the back of my mind.

> Can you also share the info from te app on a website or social media, live overview?

I mean, it's possible, but why? Tell me more about this, what would you like to share ?

FYr76, 2020-04-13:
Maybe this link is useful: https://github.com/vrachieru/plant-database

emericg, 2020-04-14:
Yes it would be very much useful if it was legal ^^ unfortunately I highly doubt it is...

minosrex, 2020-05-08:
Hi,

I also bought a few VegTrug sensors. This app came recommended with them, so I installed it.

I have to say that, so far, it works brilliantly. In fact, the sensor reading speed is much faster than the official VegTrug app, and the interface is much nicer! So very good job! 

The only advantage of VegTrug is indeed its database: I bought these sensors assuming that watering and fertiliser requirements for, say, my olive tree and lavender plants would be very different to those of my roses and, not being a good gardener, the app would tell me what to do for which plant. So while Vegtrug is technically and visually inferior, it does tell me more of what I need to know. This being said, I have to add that a) the VT database only found one of the three plants I actually use it for right now (and it's not like a Dog Rose is an uncommon plant in Europe), and that the limits for these three plans (I used approximates for the other two) aren't that far apart. But I could assume that, for example for mediterranean herbs and plants, this would be quite different. So yes, coupling this to a database would be a HUGE plus!

Best regards,
Simon.

emericg, 2020-05-13:
Hi,
Thank you for the kind words.

> I also bought a few VegTrug sensors. This app came recommended with them, so I installed it.

What do you mean by that, where did you find the recommandation?

> The only advantage of VegTrug is indeed its database

I very much agree, but I'm in the same situation as you, I'm not a gardener ^^ I took the values I needed from my time with the official app.
I can link the application to a database but the problem remain, is there an open plant database anywhere on the internet?

minosrex, 2020-05-18:
> Reply to: emericg/WatchFlower <reply@reply.github.com>

https://www.xiaomiproducts.nl/xiaomi-vegtrug-flora-plant-sensor.html

It says: “De sensor is ook goed te gebruiken met de third party app WatchFlower.” (The sensor also works fine with the third party app WatchFlower”. Nice eh? 😊

 

Plant Databases -not easy, especially since, well, I’m no specialist 😊

The best I could find so far is the USDA Plant Database. It has characteristics records like this one: https://plants.sc.egov.usda.gov/java/charProfile?symbol=CAAP2

Especially the “growth requirements” could be very useful if translated into values by a specialist. I read there is an open access DB here: https://www.gbif.org/dataset/705922f7-5ba5-49ab-a75d-722e3090e690#dataDescription. But I would not know how to use it, maybe you do?

 

For crops, there is this: https://github.com/openfarmcc/OpenFarm

 

Then, there are interesting websites containing soil and moisture info but they seem to be web interfaces, not open datasets? Eg (I took the example page for Wisteria Sinensis which I have)

PFAF: https://pfaf.org/user/Plant.aspx?LatinName=Wisteria+sinensis

Garden.org plants database: https://garden.org/plants/view/79585/Chinese-Wisteria-Wisteria-sinensis/

Practical plants: https://practicalplants.org/wiki/Wisteria_sinensis (here the top icons on light, water, soil and hardiness could be enough if they can be ‘harvested”’ (hah) and translated into values)

 

That’s what I have so far…

Simon

minosrex, 2020-05-26:
Just a question. I bought a few more sensors so now I have 14. I listed all the plants in the app, but now two things:
- How can I change the order of the plants so I can group them according to location (and make sure my smartphone is close enough for the updates). Now I find myself moving back and forth between the different groups of plants;
- I added the sensors on the OSX app as well. I wanted to use the MAC addresses to quickly find which one is which, but the OSX app and the android app show different MAC addresses. Any clues on how I can do this (without having to check which sensors have the same values, which will be much more tedious)

Thanks!
Simon.

emericg, 2020-05-27:
> https://www.xiaomiproducts.nl/xiaomi-vegtrug-flora-plant-sensor.html
> It says: “De sensor is ook goed te gebruiken met de third party app WatchFlower.” (The sensor also works fine with the third party app WatchFlower”. Nice eh?

Yeah that's pretty nice ^^

> Plant Databases -not easy, especially since, well, I’m no specialist 

Thanks for the links these are good resources, especially Practical plants (but it's a wiki).

> How can I change the order of the plants so I can group them according to location

I actually never though about that. Grouping (or at least ordering them) by location is a great idea (and easy to implement) I will do that for the next version.
(That is a lot of sensors btw)

> but the OSX app and the android app show different MAC addresses

Unfortunately macOS and iOS are using some form of UUIDs for Bluetooth devices while linux/android/windows/... are simply using devices mac addresses.
At some point I want to implement the LED blink but until that sorry :p

emericg, 2020-07-06:
@minosrex The new beta released today has both the features you asked, order by location/water level/device type and LED blink to identify devices. Try it if you can or wait a few more days for public release :)

minosrex, 2020-07-06:
Excellent! Looking forward to it. Forgive me, I'm a Github newbie, how do I get betas? (and make sure I don't lose my data :-))

FYr76, 2020-07-06:
LED bink works great, love the outside mode!
• Don't understand the light bulb icon next to refresh icon.
• The settings menu becomes a bit crowded

emericg, 2020-07-07:
> Excellent! Looking forward to it. Forgive me, I'm a Github newbie, how do I get betas? (and make sure I don't lose my data :-))

Yes I should have mentioned, it's an Android beta. You need to go to the play store, watchflower, detailed infos, and there should be a "join beta" button somewhere.

https://support.google.com/googleplay/answer/7003180

> • Don't understand the light bulb icon next to refresh icon.
> • The settings menu becomes a bit crowded

I'll do a little "three dots" menu to have refresh, led blink and sync history icons + texts in the next version.
I'll also remove a few settings I think, or maybe do an advanced settings toggle.