emericg/WatchFlower #6: Scale and threshold values

Megachip, 2020-04-07:
Is there a reason, why scale and the threshold values are much lower than they are provided by the sensor (and even by the plant db?)
E.g. for the plant showing in the screenshots, the plant db values are:
soil moisture: 22-70 
soil fertility: 200 - 2500 
light intensity: 3500 - 120000
temperature: 10 - 35

[image]
[image]

Megachip, 2020-04-07:
> Is there a reason, why scale and the possible threshold values are much lower than they are provided by the sensor (and even by the plant db?)
> E.g. for the plant showing in the screenshots, the plant db values are:
> soil moisture: 22-70
> soil fertility: 200 - 2500
> light intensity: 3500 - 120000
> temperature: 10 - 35
> 
> [image]
> [image]

emericg, 2020-04-07:
No honestly this has been done over time as I was feeling it ^^
The indicators are set to cover general use cases, from 0, to "not too high" values (as an example, if your plant is always between 10/20% humidity and the max is 70, the bar would always feel empty).
The values are fixed to give a consistency between multiple sensors, but that's up for debate.
If you have remarks let me know I'm very open to improving this!

- Moisture has 50% max because more mostly means "just watered" anyway.
- Temperature is set to 0/40 because that's (mostly) the range of our planet ^^ I have one sensor on my balcony and it does vary from one extrem to the other. Even inside I go from 18 to 38.
- The idea with luminosity is that over 10k, it means sunny anyway. Maybe this bar could be logarithmic... Also there could be a warning after 8k for plants that don't like direct sunlight.
- Fertility is a bit hard to control, in my experience it's only high when wet (but I was thinking about changing the max to 1000). Also I don't think I ever saw a value higher than 1000.

Megachip, 2020-04-08:
Ok, that's for thresholds ... but what for scaling? If all bars reaching the max (or better are far right of the scale end), whats the sense of showing this? What about dynamic scaling? So at least the max value (all time? last month) will be on the scale... And it would fit for every sensor.

emericg, 2020-04-08:
> If all bars reaching the measured max (or better are far right of the scale end), whats the sense of showing this?

Well right now it probably means your plant is good ^^ otherwise you'll have little red warnings showing (if you have too much water or too high temp).

> What about dynamic scaling? So at least the max value (all time? last month) will be on the scale... And it would fit for every sensor.

Yes that can definitely be done. It's already used by the graphs. I'll try this and see if it helps getting a better sense of the high values and doesn't kill readability of low values.

emericg, 2020-04-14:
What platform are you using? And do you build the software yourself?
If yes, with the latest commit you can just set the indicatorsDynascale value to true to test the dynamic scales.

Megachip, 2020-04-15:
> What platform are you using? 
Android
> And do you build the software yourself?
Not yet. 
> If yes, with the latest commit you can just set the indicatorsDynascale value to true to test the dynamic scales.
I'll guess my android SDK and Build Tools are some years old. But if I find the time, I'll check it.

emericg, 2020-04-15:
> I'll guess my android SDK and Build Tools are some years old. But if I find the time, I'll check it.

Don't worry about it, I can send you a beta version instead. Plus it's a Qt software, it's more involved to build than a regular android application.

I'll need to think about it some more but right now it's set up like this, how does that sound ?
- 0 to hygroMax*1.10 for moisture.
- tempMin*0.80 to tempMax*1.20 for temperature.
- 0 to lumiMax*1.10 for luminosity.
- 0 to conduMax*1.10 for conductivity.
It's using the 30 days history, not the 14 of the graph.

Megachip, 2020-04-15:
> I'll need to think about it some more but right now it's set up like this, how does that sound ?
> 
> * 0 to hygroMax x 1.10 for moisture
> * tempMin x 0.80 to tempMax x 1.20 for temperature
> * 0 to lumiMax x 1.10 for luminosity
> * 0 to conduMax x 1.10 for conductivity
>   It's using the 30 days history, not the 14 of the graph.
Sounds perfect. If you like to have it tested, feel free to provide an APK. Maybe also including the latest translations. So we can test this too.

Thanks a lot.

emericg, 2020-04-19:
OK the beta version is up https://play.google.com/apps/testing/com.emeric.watchflower 
Let me know how you feel about the dynamic scale!
Also if you want to make adjustments to the translation don't forget to pull the latest version first ;-)

emericg, 2020-04-21:
Beta 3 is up, last chance to have a look at this ^^

emericg, 2020-04-25:
I'm closing this, feel free to comment or reopen if something comes up.

closed 2020-04-25 by emericg