emericg/WatchFlower #7: Translating

Megachip, 2020-04-07:
Any guide how to help on translation? Or just take one of the TS files?

emericg, 2020-04-07:
Hi, what language would you like to add? I can just push a new empty TS file for you.
If it's an existing language yes you can just open the TS file with QtLinguist or Poedit and push the results!
Let me know if you have other questions.

Megachip, 2020-04-08:
> Hi, what language would you like to add? I can just push a new empty TS file for you.
German

emericg, 2020-04-08:
I pushed a German file, you can modify it directly.
When finished you can use the "file, release"  function to make a qm file, that's the one that the software will use. Thanks!

Megachip, 2020-04-09:
> Hi, what language would you like to add? I can just push a new empty TS file for you.
> If it's an existing language yes you can just open the TS file with QtLinguist or Poedit and push the results!

Poedit (https://github.com/vslavik/poedit) seems not understand the Markup nor the binary file ... will try QtLinguist. 

Maybe you should use web services like www.transifex.com (not sure, if it is still free, but know it from other OS projects), which not requires external tools.

emericg, 2020-04-09:
Damn, the ts files can be converted to po files that's why I was able to use poedit the other day.
Transifex use to be a solution for open source projects but now you need to email them and make your case and hope they approve your, it just suck (otherwise it cost thousands per year).
I tried to find an online platform for translation two weeks ago, and found none remotely usable unfortunately...

Megachip, 2020-04-09:
Will it fall back to English, if a string is missing? Or must the translation 100% completed?

emericg, 2020-04-09:
Yes it will fallback to english. Translate as much as you want, leave the rest ;-)

emericg, 2020-04-12:
Thank you for the German translation it's awesome!

I created a quick translation guide in i18n/README.md, I'll improve it as much as possible in the futur.
I've added credit to the growing list of translators in this file too, if you want to provide another pseudo/name and email/github page, feel free to mention it here.
At some point I should also put that list directly into the application.

closed 2020-04-12 by emericg

emericg, 2020-04-15:
Also, if you'll be so kind, can you translate these sentences (at least the first one) so I could put a basic German translation for the iOS and Android store? That would help indicating that German is a supported langage.

```
A plant monitoring application for Xiaomi "Flower Care" and "RoPot" Bluetooth sensors and thermometers!

WatchFlower is a plant monitoring application that reads and plots data from your Xiaomi "Flower Care" and "RoPot" Bluetooth sensors. WatchFlower also works great with a couple of Xiaomi Bluetooth thermometers!

WatchFlower is an open source software. By buying it here, you support its development!
It doesn't track the usage you have of the application, doesn't require an account creation, your GPS location, nor any other personal data from you!
```

Megachip, 2020-04-15:
> A plant monitoring application for Xiaomi "Flower Care" and "RoPot" Bluetooth sensors and thermometers!

I'll guess I did?
[image]
[image]
```
WatchFlower ist eine Open Source Software. Indem Sie es kaufen, unterstützen Sie die Entwicklung!`
Es verfolgt nicht Ihr Nutzungsverhalten, erfordert keine Registrierung, verfolgt nicht Ihren Standort oder zeichnet andere persönliche Daten von Ihnen auf.
```

emericg, 2020-04-19:
Thank you so much for the new translation.