emericg/WatchFlower #10: Dutch translation

FYr76, 2020-04-12:
Hi there,

Your app is growing more and more and starting to flower. Love it!
I'd like to contribute with a Dutch translation. I'm compactly new to contributing on GitHub, so I hope you can help me a bit with the necessary steps.

Kind regards,  FYr76

FYr76, 2020-04-12:
If easy to do simultaneously, I could also translate into Frisian

emericg, 2020-04-12:
Hi! Yes translations are very welcomed.

I've added Dutch and Frisian translation files into the application, as well as a quick guide on how to add translations (https://github.com/emericg/WatchFlower/blob/master/i18n/README.md).
You can add Frisian too, but you'll see that one language already take a fair amount of time to do!

You'll need to open the corresponding .ts files with Qt Linguist (https://github.com/lelegard/qtlinguist-installers/releases).
But before that, you should fork the repository, create a new branch, open the translations file there, translate/save and then create a commit, then a pull request.
Here is a great step by step guide from GitHub: https://github.com/firstcontributions/first-contributions

You can also use a graphical client to deal with all of these steps, like GitKraken or Sourcetree.
Let me know if you need help.

FYr76, 2020-04-13:
Bit more complicated than I thought.  I don't have a desktop at home, so I started to translate/ edit the .ts file itself. Is it okay  to do it that way?

emericg, 2020-04-13:
Yeah sure, the translation software helps with a couple subtleties but it's doable without!

FYr76, 2020-04-13:
Thanks. I'll keep you updated

FYr76, 2020-04-13:
I have to translate 'moisture' and 'humidity' in seperate lines. What's the difference I have to take account off? Perhaps soil moisture  and air humidity?

emericg, 2020-04-13:
Yes it is soil moisture and air humidity. Some languages have different words, others don't.
The application supports plant sensors and thermometers so it make sense having both.

FYr76, 2020-04-13:
462 <source>at</source

What is the context of the word  'at' there?

emericg, 2020-04-13:
It's for these date/time indicators at the center of the screen.

[image]

FYr76, 2020-04-13:
I understand 😃 
Almost finished

FYr76, 2020-04-13:
484 <translation>Weet je zeker dat je gekozen apara(a)t(en) wilt verwijderen?</translation>
Translating devices) by apara(a)t(en) is ugly, is it okay to use sensor(s).... same word in Dutch

583 <source>Official informations</source>
What do you mean here?

611 <source>Refresh sensors data</source>
Is it one or more devices?

680 <source>Change persistent settings here!</source>
What do you mean here? Predefined?

690 <source>Automatique dark mode</source>
Automatic is correct English

705 <source>Bluetooth control</source>
+ 720 
What exactly do you mean?

710 <source>WatchFlower can power on your device&apos;s Bluetooth in order to operate.</source>
I translated it like: WatchFlower can activate your device's Bleutooth. 
Is that adequate?

715         <source>Bluetooth compatibility</source>
What do you mean here?

720 source>Sensors will be updated sequentially instead of simultaneously. Improve Bluetooth communication reliability, at the expanse of synchronization speed.</source>
What do you mean with the 2nd sentence?

725 <source>Start application minimized</source>
Do you mean pop up or small app-windows?

740 what do you mean with        system tray?

767 what is        h.?

772 source>Use bigger indicators</source>
Do you mean labels?

787 source>Monthly</source>
+ 792
Is this about the histogram? 

870 <source>Click on sensors to access
Click on one of the sensors?

880 <source>Set &lt;b&gt;limits&lt;/b&gt; like &lt;b&gt;optimal water le&lt;/b&gt; or &lt;b&gt;temperature range&lt;/b&gt; and more depending on available sensors metrics.</source>;
Optimal moisture level as basis for translation?

and more depending on available sensors metrics: is     depending on sensors specifications       okay as basis for translation? 

885 source>Previous</source>
890 source>Allright!</source>
What's the context?

emericg, 2020-04-14:
> 484 Weet je zeker dat je gekozen apara(a)t(en) wilt verwijderen?
> Translating devices) by apara(a)t(en) is ugly, is it okay to use sensor(s).... same word in Dutch

Sure.
In any case, when you're not sure about a translation, leave the 'unfinished' tag, it will make it easy for other people to have a second look at it.

> 583 Official informations
> What do you mean here?

It's in a Bluetooth information text. It's a link to an official android documentation page.

> 611 Refresh sensors data
> Is it one or more devices?

It's in the app drawer, to refresh all sensors at once.

> 680 Change persistent settings here!
> What do you mean here? Predefined?

For the desktop version. It's for settings that are saved.

> 690 Automatique dark mode
> Automatic is correct English

Damn you're right, Automatique is french ^^ Sometimes I let english slip into my french and vice versa...

> 705 Bluetooth control
> What exactly do you mean?

That's the title for the setting below.

> 710 WatchFlower can power on your device's Bluetooth in order to operate.
> I translated it like: WatchFlower can activate your device's Bleutooth.
> Is that adequate?

Yes indeed, I should probably use that too

> 715 Bluetooth compatibility
> What do you mean here?

That's the title for the setting below. That was really hard to put into word...

> 720 source>Sensors will be updated sequentially instead of simultaneously. Improve Bluetooth communication reliability, at the expanse of synchronization speed.
> What do you mean with the 2nd sentence?

With many android phones, bluetooth usually sucks hard. This settings helps improving reliability, but instead of updating each sensors on a 10s period, il will take 10s * the number of sensors. So if you have 10 sensors... that's why I put "at the expanse of synchronization speed."

> 725 Start application minimized
> Do you mean pop up or small app-windows?

For the desktop version. The application starts (like when the computer starts) but it's hidden/minimized on the task bar.

> 740 what do you mean with system tray?

For the desktop version. For instance, the bottom/right area of the screen on windows with the little icons for applications that stays always on. This thing doesn't necessarily have a straightforward translation. In french I thing I use "notification area" or something.

> 767 what is h.?

Short for hourly, if applicable in dutch.

> 772 source>Use bigger indicators
> Do you mean labels?

> 787 source>Monthly
> 792
> Is this about the histogram?

Yes

> 870 Click on sensors to access
> Click on one of the sensors?

Yes It's better indeed

> 880 Set <b>limits</b> like <b>optimal water le</b> or <b>temperature range</b> and more depending on available sensors metrics.;
> Optimal moisture level as basis for translation?

yes indeed

> and more depending on available sensors metrics: is depending on sensors specifications okay as basis for translation?

I guess I meant more "depending on data mesured by sensors", like some sensors mesure light, some moisture and light, etc...

> 885 source>Previous
> 890 source>Allright!
> What's the context?

It's from the tutorial, to go between the pages. You can see it in about / tutorial.

FYr76, 2020-04-14:
Thank you. Good to know.

I'm finished translating. What's the next step?

By the wat,  is it possible to add lines that give info about the app text? For example the context, the corresponding device, including maximum caracters? Guidelines could make it easier to translate.

emericg, 2020-04-15:
> What's the next step?

You should be able to start a pull request like this : https://github.com/firstcontributions/first-contributions/blob/master/README.md#submit-your-changes-for-review

I'll merge the changes, check the file format (because you edited it manually it's probable some manual fixes will be required) and we'll be good to go!
Tell me if you want more than your pseudo added to the translators list in i18n/README!

> By the wat, is it possible to add lines that give info about the app text? For example the context, the corresponding device, including maximum caracters? Guidelines could make it easier to translate.

I think it is possible to add comments yes, but I never did it yet. I'll see if I can add some, based on your remarks.
The translation software helps put the strings in context by showing the code around. Also you can add comments, per translation file, not globally, to help futur translation or explain your choices.
Regarding word and sentences sizes, there just isn't any definitive answer to that problem. Sometimes the application can adapt sometimes too much characters breaks the UI... Trial and error...

Also, if you'll be so kind, can you translate these sentences (at least the first one) so I could put a basic Dutch translation for the iOS and Android store? That would help indicating that Dutch is a supported langage.

```
A plant monitoring application for Xiaomi "Flower Care" and "RoPot" Bluetooth sensors and thermometers!

WatchFlower is a plant monitoring application that reads and plots data from your Xiaomi "Flower Care" and "RoPot" Bluetooth sensors. WatchFlower also works great with a couple of Xiaomi Bluetooth thermometers!

WatchFlower is an open source software. By buying it here, you support its development!
It doesn't track the usage you have of the application, doesn't require an account creation, your GPS location, nor any other personal data from you!
```

FYr76, 2020-04-15:
I'm going to read the link and pull a request.
Later I'm going to do the Frisian version. Nice idea to add a Dutch text to the app stores. No need to have a separate Frisian version as it's a minority language in a part of The Netherlands.

Only my pseudo is fine enough.

You say I can add comments per translation file. Is that in the box above the commit button?

Gonna work on the text above right now

FYr76, 2020-04-15:
A plant monitoring application for Xiaomi "Flower Care" and "RoPot" Bluetooth sensors and thermometers!
Een plantobservatieapp voor Xiaomi "Flower Care" en "RoPot" bluetoothsensoren en thermometers!
* You chose for RoPot, that's different from Ropot in the app 
* what is the purpose of the exclamation mark, is it to emphasize the whole sentence or the fact its also for thermometers? I need to know for the translation

WatchFlower is a plant monitoring application that reads and plots data from your Xiaomi "Flower Care" and "RoPot" Bluetooth sensors. WatchFlower also works great with a couple of Xiaomi Bluetooth thermometers!
WatchFlower is een plantobservatieapp die data van je Xiaomi "Flower Care" en "Ropot" bluetoothsensoren leest en overzichtelijk weergeeft in diagrammen. WatchFlower werkt ook goed met een aantal Xiaomi bluetooth-thermometers!
* I used this a basis : WatchFlower is a plant monitoring application that reads data from your xxxx and shows it in clear charts......... Okay with you?

WatchFlower is an open source software. By buying it here, you support its development!
WatchFlower is open source software. Door het hier te kopen steun je verdere ontwikkelling van de app!

It doesn't track the usage you have of the application, doesn't require an account creation, your GPS location, nor any other personal data from you!
Het volgt (Wij volgen) je appgebruik niet, een account aanmaken is niet nodig, evenmin als  je GPS-locatie of andere persoonsgegevens.
* is this better? I used it as basis. I prefer We btw...   :      It doesn't (We don't) track your app usage, account creation isn't requierd, nor your GPS location or any other personal data!

to translate the following line in Frisian (that problem doesn't exist in Dutch) I need to know if it's hour, hours or hour(s)
94    <source> hours</source>

FYr76, 2020-04-15:
788 793 monthly and weekly has to be translated like that and not just as month and week?

FYr76, 2020-04-15:
Both languages ready now

FYr76, 2020-04-16:
Perhaps you could mention the added changes in the "New functions section" of the Playstore:

Added languages:
Dutch
Frisian

Added languages:
Dutch (Nederlands)
Frisian (Frysk)

 Or

Added languages
Nederlands
Frysk

I prefer the last

emericg, 2020-04-16:
You are going too fast for me ^^

> I'm going to read the link and pull a request.

It should be just one click to start a pull request. One pull request per language would be better.

> You chose for RoPot, that's different from Ropot in the app

I'm changing it everywhere

> what is the purpose of the exclamation mark, is it to emphasize the whole sentence or the fact it also supports thermometers? I need to know for the translation.

whole sentence I guess

> to translate the following line in Frisian (that problem doesn't exist in Dutch) I need to know if it's hour, hours or hour(s)

I thinks it's hours. Usually if the quantity can be either singular or plurar, I set it up like hour(s) and then the translation software offer to handle specifics depending on the target language https://doc.qt.io/qt-5/i18n-source-translation.html#handling-plurals

> has it to be translated like that (per month/ week) or is it rather month and week?

It's for monthly and weekly view in the histograms, but I don't know if one is really better than the other, feel free to improvise :)

> Perhaps you could mention the added changes in the "New functions section" of the Playstore:

Yes of course ^^

FYr76, 2020-04-16:
I pulled a request. Hope is was the right way. Give me a sign when I can pull the Frisian translation. It's ready too.

emericg, 2020-04-18:
Are you using an Android device? If yes I'll wait for your next pull request to do a beta release so you can have a look at the result!

FYr76, 2020-04-18:
Yeah, I have Android here. Curious about how it with works out. 
I merged the  pull request with my rep. Trying it again...

emericg, 2020-04-18:
Yes good job the new commits got added to the PR.
I've check the file they're basically good to go, very few adjustments needed. For a "manual" translation that was very well done! Thank you so much for this work!

I'm gonna play battlefield now but I'll do the BETA tomorrow ^^

FYr76, 2020-04-19:
I gave it a thought: maybe this way it's clearer what monthly and weekly is about

**  782        <source>Histogram overview</source> **

emericg, 2020-04-19:
OK the beta version is up https://play.google.com/apps/testing/com.emeric.watchflower

FYr76, 2020-04-19:
My evaluation here?

emericg, 2020-04-19:
As you wish, you can always send me an email if you want.

FYr76, 2020-04-19:
ABOUT SECTION
----

### website link
- proposal: translation of the website
### github page link 
- Proposal: translation of the website
### support link 
- proposal: translation of the website

### translation correction
- 49  <translation>Beoordeel deze app</translation>

APP SETTINGS
----

- automatic dark mode (isn't translated)
- WatchFlower can activate your device's ....   (isn't translated)- - 
- Dynamic scale for indicators (isn't translated)

_I don't understand this funcion_

### text correction
- 720   at the expanse of must be  :  at the expense of (or at the cost of)

### text suggestion
- 720   <source>Sensors will be updated sequentially instead of simultaneously. This improves Bluetooth communication reliability, but at the expense of synchronization speed.</source>
- 720   <source>Sensors will be updated sequentially instead of simultaneously. This improves Bluetooth communication reliability while synchronization only takes a bit longer.</source>

_the last one doesn't emphasize on the negative aspect of loss of sync speed.... it's more positive_

### translation correction
721  <translation>De sensoren worden na elkaar geüpdatet in plaats van gelijktijdig. Dit verbetert de betrouwbaarheid van de bluetoothverbinding terwijl de synchronisatie maar iets lager duurt.</translation>

### switching languages
- there's no toggle for switching languages

### text suggestion
772  <source>Use bigger descriptions</source>
772  <source>Use bigger data labels</source>

### translation correction
773   <translation>Gebruik grotere data labels</translation>

SENSOR OVERVIEW SECTION
----

- bodemvochtigheid (can only see `demvochtigheid` on Tab S4, rest outside margines)
- bodemvruchtbaarheid (can only see `vruchtbaarheid` on Tab S4, rest outside margines)

- when there's no other option you can replace them with `vochtigheid`  en `vruchtbaarheid` 

### firmware update
I see two older devices need a firmware update  
   <source>Update available!</source>

* it's not clear `Update available!` is clickable 
* it's not clear which app you mean
* maybe you could suggest if it's necessary / wise to upgrade (I heard about breaking the sensor because of upgrade errors). 
* You could give some privacy and pairing suggestions (specially with the Chinese sensors)

SENSOR OVERVIEW / HISTOGRAM SECTION
----

- 782  <source>Histogram</source>
- It is okay like this (I suggested 'Histogram overview' earlier today)

SENSOR OVERVIEW / HISTORY SECTION
----

### week overview 
- the abbreviations of the days aren't correct, it's supposed to be: `ma. di. wo. do. vr. za. zo.`

### month overview 
- correct abbreviations of months are: `jan. feb. mrt. apr. mei jun. jul. aug. sept. okt. nov. dec.` (mei is without period)

### suggestion
- add abbreviations to ts translation files

SENSOR OVERVIEW / SETTINGS
----

### translation correction
- 330   <translation>Sommige planten staan graag de hele of een gedeelte van de dag in de volle zon. Maar de meeste kamerplanten hebben liever geen direct zonlicht. Plaats ze daarom niet voor een venster op het zuiden!</translation>
- 340   <translation>Het vruchtbaarheidsgehalte van de grond is een indicatie voor de beschikbaarheid van voedingsstoffen in de grond. Gebruik meststoffen (met mate) om de voedingswaarde op peil te houden.</translation>

### specifying name and location 
- Once a name and location is specified one can't see how to change it or correct it at a later moment. 
- maybe you could place a pictogram of a pencil next to location and name

GENERAL STUFF
----

### Luminosity
- 309   <source>low</source>   (low isn't consistant with indirect, direct and sunlight)

### translation correction
- 471  <translation>min.</translation>
- 476  <translation>max.</translation>

### Translation question 
I translated application everywhere with app... should it be different (application instead of app) at some places where it is about Windows or Linux?

TUTORIAL
----

### translation missing
- 871  <translation>Klik op een sensor voor &lt;b&gt;gedetailleerde info&lt;/b&gt;, &lt;b&gt;grafieken&lt;/b&gt; en &lt;b&gt;historische data&lt;/b&gt;.</translation>

### translation correction
- 876   <translation>Je kunt je &lt;b&gt;plant labelen &lt;/b&gt; en de &lt;b&gt;locatie&lt;/b&gt; van de sensor aangeven.</translation>

### translation missing
- 881   

### translation correction
- 881   <translation>Stel &lt;b&gt;grenswaarden&lt;/b&gt; in zoals &lt;b&gt;optimale vochtigheidsgraad&lt;/b&gt;, &lt;b&gt;temperatuurbereik&lt;/b&gt; en meer afhankelijk van de sensorspecificaties.</translation>

### Translation question 
- 891  <source>Allright!</source>
- wrong spelling? https://www.merriam-webster.com/words-at-play/all-right-or-alright-which-is-correct
- This can't be translated in this context. My `Zie zo!` isn't nice too.
- I suggest using an alternative word that better suits the context, both in English and Dutch.
- Can I use `Understood?`, `Ready?`, `Ready!` or just `Close` since it has a closing function?

emericg, 2020-04-20:
OK I think I fixed most of the translations.

> proposal: multilingual website / readme

just way too much work, it would makes everything harder to evolve and maintain, sorry

> doesn't emphasize as much on the negative aspect of loss of sync speed.... it's more positive

I see what you mean, but it doesn't make it a little slower, it can make it 10x slower if you have 10 sensors

> there's no toggle for switching languages

That's on purpose, on mobile the applications usually just take the langage of the device

> headers

I agree, current presentation is not ideal

> SENSOR OVERVIEW SECTION

I think I'll use the short form then, I know that the label size is a problem but it's actually not possible to fix this without rewriting this entire section. But at some point a solution must be found.

> firmware update

I just removed the update indicators. I can't tell if it's wise to upgrade or if it's even useful anyway.

> You could give some privacy and pairing suggestions

That's only needed for the official application. This one works entirely offline and with any device version.

> Histogram overview

It's more "histogram default content to be showned", not much overview. Also on phone or small screen it's best to keep this one short or it will overlap with the spinbox

> week / month overview

What's wrong with them? These translations came from Qt directly I can't change them.

> add abbreviations to ts translation files

Yes I'll try to in the futur

> specifying name and location

I see what you mean, on desktop the pencil appears whenever we hover the mouse so it's quite obvious. I'll make it permanent on mobile.

> when changing one of the parameters Gesynchroniseerd turns orange

When you update the interval? Orange mean that the data sync is old (not between 0 and the update interval) but still ok (less than 12h old).

> when selecting the fertility limits to the max. I only see a 1 instead of 1000

indeed I saw this one but forgot to fix it

> should it be different (application instead of app) 

Yes it probably should be application everywhere. But that's not very important. App is just Apple and Google's cute way to make softwares seem fun.

> adding : the flashlight lights up after succesfully pairing the device.

This would be cool but it's really not that easy to do. Adding the capability to make the led flash is on a todolist somewhere though.

> Can I use Understood?, Ready?, Ready! or just Close since it has a closing function?

Yes you can use the one you like. It's a similar case to your remark on the low/indirect/sunlight. Even on a small software you got so many strings to come up with, at some point you just start to wing it ^^ Most of the time there is no right or wrong, you just got to be as close as possible to what you mean.
Like the orange color to "synced" to mean that the update is "good, but old", it may not be ideal but it's a straightforward solution to a problem that we could spend a day discussing.

> On the overview and on the histograms you could add some Y-axis for readability

I see what you mean but not that easy, and every attempt to make a readable and pretty Y-axis has failed so far

FYr76, 2020-04-20:
Thanks for spendings time on my comments.

### translating website, GitHub page and Support page
You're right, too much work and not first priority

### line 720
It realy takes much longer than I thought. In that case line 721 has to be changed.

### translation correction
721 De sensoren worden na elkaar geüpdatet in plaats van gelijktijdig. Hoewel dit ten goede komt aan de betrouwbaarheid van de bluetoothverbinding zal synchroniseren langer gaan duren.

### switching languages
I understand the application usually takes the language of the device. Minority languages on the other hand (like Frisian, Basque or Ch'ti) usually can't be set as the main language on a device. I would really love to use my first language in WatchFlower. Mybe you can reconsider. 

### week / month overview
In the month overview in the histogram I see `apr.` as abbreviation of `april`. That's correct. I don't know how the other months are abbreviated, that's why I wrote them down for you to look at it.

In the week overview of the histogram the weeksday's are abbreviated to 3 caracters. It's not wrong, but as` Sunday / zondag` is always shortend to `zo.`, keeping all to 2 caracters gives a more consistent look.

Pitty you can't change them :)

### Synced in orange
Synced turned orange after I changed the one of the settings of the individual sensors. Clear explanaition of why you chose for this setting...

### adding : the flashlight lights up after succesfully pairing the device.
I meant adding the text above. When I pair a sensor the flashligt blinks a few times. Or is this a 'original app' only feature?

### translation correction
892 Sluiten

### update Frisian
I still have to update the Frisian translations based on my two last posts here... Can you give me a sign when you're finished correcting the Dutch ts-file? After that I can correct the Frisian file and do a new pull request... is that okay with you?

emericg, 2020-04-20:
> I would really love to use my first language in WatchFlower. Mybe you can reconsider.

I mean sure, that seems to be a valid reason. Right now I don't have any code to do that though so I'll have a look at it.

> the flashlight lights up after succesfully pairing the device

They made the light flash manually as far as I know yes.

I fixed the two translations you mentionned. I did a second beta before that this afternoon.

For the Frisian you can go ahead but maybe download the frisian file from this repository and mail it back. Otherwise you need to sync up the new changes, and I'm not sure how you can do that from the website.

FYr76, 2020-04-20:
translation corrections

        <translation>Wurdearje dizze app</translation>
        <translation>Keamerplanten steane leafst yn potgrûn mei in fochtpersintaazje fan 15 oant 50%. Tropyske planten steane graach wieter, fetplanten graach drûger. Dêr mei it sakje oant 7%.</translation>
        <translation>Guon planten steane graach de hiele of in part fan de dei yn 'e folle sinne, mar de measte keamerplanten hawwe lykwols leaver gjin direkt sinneljocht. Tink dêr om foarsto planten foar in rút op it súden setst!</translation>
        <translation>min.</translation>
        <translation>maks.</translation>
        <translation>Yn Android 6+ is lokaasje-machtiging ferplichte foar it skennen nei Bluetooth Low Energy apparaten. De app brûkt dyn lokaasje net en slacht dizze allikemin op. Ekskús foar it ûngemak.</translation>
        <translation>De sensoaren wurde neiïnoar bywurke ynstee fan tagelyk.Hoewol't dit  te'n goede komt oan de betrouberens fan de bluetoothferbining, sil it bywurkjen al langer duorje.</translation>
        <translation>Brûk gruttere data labels</translation>
        <translation>Do kinst dyn &lt;b&gt;plant labelje&lt;/b&gt; en de &lt;b&gt;lokaasje&lt;/b&gt; fan de sensor oanjaan.</translation>
        <translation>Jou de &lt;b&gt;grinswearden&lt;/b&gt; op, lykas &lt;b&gt;optimale fochtigensgraad&lt;/b&gt;, &lt;b&gt;temperatuerberik&lt;/b&gt; en mear ôfhinklik fan de sensorspesifikaasjes.</translation>
        <translation>Slute</translation>

on the first screenshot above you shortened the words bodemvochtigheid and bodemvruchtbaarheid.   
The same can be done in Frisian. I don't know on which lines
Luchtfochtigens can be shortened to fochtigens
Grûnruchtberens can be shortened to fruchtberens

You prefer application above app. In that case it's okay to replace app with application

in Dutch that wil be: applicatie
in Frisian that wil be: applikaasje

FYr76, 2020-04-20:
Great, downloaded the beta. It's almost perfect.

[image]

### 1. You preferred application . At this screenshot it should be: 

Dutch
```
Applicatiethema
```
 
Frisian
```
Applikaasjetema
```
### 2. I don't see the last correction. 

Dutch
```
De sensoren worden na elkaar geüpdatet in plaats van gelijktijdig. Hoewel dit ten goede komt aan de betrouwbaarheid van de bluetoothverbinding zal synchroniseren langer gaan duren. 
```
  
Frisian
```
De sensoaren wurde neiïnoar bywurke ynstee fan tagelyk.Hoewol't dit te'n goede komt oan de betrouberens fan de bluetoothferbining, sil it bywurkjen al langer duorje.
```
### 3. Translation 

Dutch
```
Dynamische schaalverdeling 
```
Frisian
```
Dynamyske skaalferdeling
```
[image]

### 1 and 2 are corrections
Dutch
```
Max.
Min.
```
Frisian
```
Maks.
Min.
```
### 3. Margins
Dutch
```
Text fits very tightly
```

[image]

### replace text
Dutch
```
Sluiten
```
Frisian
```
Slute
```

emericg, 2020-04-21:
> I don't see the last correction

Beta 2 didn't have them yet. Beta 3 is up, and it will probably be the last.

> Max. / Min.

The min/max are lower case and no dot on purpose, because they are displayed inside the data indicator they need to be as compact as possible

> Text fits very tightly

That would have to be improved later ^^

> Playstore text

It's still the same text that you sent the other day. I can't add the play store page before releasing the 0.12 version or it will be kind of false advertising ^^ If your motivated we can add a few more sentences in Dutch instead of English fallback.

```
Een plantobservatieapp voor Xiaomi "Flower Care" en "RoPot" bluetoothsensoren en -thermometers!

WatchFlower is een plantobservatieapp die data van je Xiaomi "Flower Care" en "Ropot" bluetoothsensoren leest en overzichtelijk weergeeft in diagrammen. WatchFlower werkt ook goed met een aantal Xiaomi bluetooth-thermometers!

WatchFlower is open source software. Door de app hier te kopen steun je verdere ontwikkelling!
(Het volgt) (Wij volgen) je appgebruik niet, een account aanmaken is niet nodig, evenmin als je GPS-locatie of andere persoonsgegevens.
```

FYr76, 2020-04-21:
Nice... I changed everything to application

> Een plantobservatieapplicatie voor Xiaomi "Flower Care" en "Ropot" bluetoothsensoren en -thermometers!

> WatchFlower is een plantobservatieapplicatie die data van je Xiaomi "Flower Care" en "Ropot" bluetoothsensoren leest en overzichtelijk weergeeft in diagrammen. WatchFlower werkt ook goed met een aantal Xiaomi bluetooth-thermometers!

> WatchFlower is open source software. Door de applicatie hier te kopen steun je verdere ontwikkelling!
Het volgt je applicatiegebruik niet, een account aanmaken is niet nodig, evenmin als je GPS-locatie of andere persoonsgegevens.

Or

> WatchFlower is open source software. Door de applicatie hier te kopen steun je verdere ontwikkelling!
We volgen je applicatiegebruik niet, een account aanmaken is niet nodig, evenmin als je GPS-locatie of andere persoonsgegevens.

Up to you which choose
- 1st is: It doesn't follow
- 2nd is: We don't follow

Of course, no problem. Just paste me the text you want me to translate

FYr76, 2020-04-22:
I began with the Danish translations. Would you be so kind to add the files for me? Shall I open a new enhancement request?

emericg, 2020-04-22:
Damn how many language do you speak ^^
I've added the Danish files to the code. I think the best (and easiest) course of action is just to delete your current fork on GitHub, and recreate it. You'll have the latest version of the translation files, and your contribution will still be commited here.

No need to create new issues. I'll look at the language toggle today, I don't know yet how much work it is.

FYr76, 2020-04-22:
Only 3 :blush:

FYr76, 2020-04-22:
What are the lines where fertility en humidity had to be shortened?

emericg, 2020-04-22:
These are the lines from ItemIndicatorsCompact and ItemIndicatorsFilled, so around 500 to 550.

FYr76, 2020-04-22:
I realized that the word `Humidity`, at the (dynamic) chart, isn't abbreviated in Dutch and Frisian yet, like `Moisture` and `Fertility`. I guess it's being shown at the same chart when applicable. Is that right?

If so, then the word will probably be to long too / longer than `Vruchtbaarheid` (which just fits on the screen).

To avoid confusion between humidity  and moisture (when abbreviated), moisture has to be changed again. It isn't the most pretty solution, but it'll due. Therefore I kept the "unfinished" tag.

Luckily in Danish the words are short enough.

### Dutch    -   short version of `Luchtvochtigheid`
        <source>Humidity</source>
        <translation type="unfinished">Luchtvocht</translation>

### Dutch   -   short version of `Bodemvochtigheid`
        <source>Moisture</source>
        <translation type="unfinished">Bodemvocht</translation>

### Frisian   -   short version of `Luchtfochtigens`
        <source>Humidity</source>
        <translation type="unfinished">Luchtfocht</translation>

### Frisian   -   short version of `Grûnfochtigens`
        <source>Moisture</source>
        <translation type="unfinished">Grûnfocht</translation>

emericg, 2020-04-24:
> I guess it's being shown at the same chart when applicable. Is that right?

Yes indeed.

I've updated the translations and also commited the language switcher. Let me know when your good with the Danish translation and I'll do another beta! The latest (for real :p)

FYr76, 2020-04-24:
Næsten færdig. I hope today

FYr76, 2020-04-24:
• translating to multiple languages makes creative, so I reconsider things. Sorry for that.

### Dutch correction
```
717        <translation>WatchFlower kan bluetooth van je apparaat inschakelen, zodat het sensordata kan synchroniseren.</translation>
```

### Frisian correction
```
712        <translation>WatchFlower kin bluetooth op dyn apparaat ynskeakelje, sa't it sensordata bywurkje kin.</translation> 
```
• Maybe you allready deleted these lines (because of the vanished label)

### Dutch correction
```
330       <translation>Sommige planten staan graag de hele of een gedeelte van de dag in de volle zon, maar de meeste kamerplanten hebben liever géén direct zonlicht. Let daar op voor je planten voor een venster op het zuiden zet!</translation>
```

### Frisian correction
```
330         <translation>Guon planten steane graach de hiele of in part fan de dei yn 'e folle sinne, mar de measte keamerplanten hawwe lykwols leaver gjin direkt sinneljocht. Tink dêr om foarsto planten foar in rút op it súden setst!</translation>
```
### Dutch correction
```
341        <translation>Het vruchtbaarheidsgehalte van de grond is een indicatie voor de beschikbaarheid van voedingsstoffen in de grond. Met meststoffen houd je de voedingswaarde op peil; gebruik de juiste hoeveelheden passend bij het jaargetijde </translation>
```

### Frisian correction
```
341        <translation>It fruchtberensgehalte fan de grûn is graadmjitter foar de beskikberens fan fiedingsstoffen yn &apos;e grûn. Brûk dongstoffen (ôfmjitten) om de fiedingswearde op peil te hâlden en hâld rekken mei de tiid fan it jier</translation>
```

### suggestion 
- add seasonal advice
```
339     <source>Soil fertility value is an indication of the availability of nutrients in the soil. With the right quantities appropriate to the season, you keep this value up.</source>
```
### Dutch correction
```
761            <translation>WatchFlower draait op de achtergrond in systeemvak en wordt op gezette tijden actief voor het bijwerken van sensordata.</translation>
```

### Frisian correction
```
742        <translation>WatchFlower draait op &apos;e eftergrûn yn it notifikaasjegebiet en wurdt op fêste tiden aktyf om sensordata te ferfarskjen.</translation>
```
###  suggestion 
- add draught advice
```
294        <source>Having constant temperature is important for indoor plants. Protect them from draught.</source
```

### Dutch correction
```
295        <translation>Kamerplanten houden van een constante temperatuur en staan niet graag op de tocht!</translation>
```

### Frisian correction
```
295        <translation>Keamerplanten hâlde fan in konstante temperatuer en fral net fan sigerij!</translation>
```

-------
Going to do the last checks with the  Danish translations now

FYr76, 2020-04-24:
- `south-facing windows`  better?
```
329        <source>Some plants like direct sun exposition, all day long or just for part of the day. But many indoor plants don't like direct sunlight: place them away from south oriented windows!</source>
```

FYr76, 2020-04-25:
It's ready, but found a stupid mistake

### Danish
<translation>Hvad vil du hvide?</translation>

FYr76, 2020-04-25:
### English
- plants instead of plant
84   <source>You need to water one of your (unnamed) plant!</source>

emericg, 2020-04-25:
Ok I published the last beta, hopefully I included most of your corrections and remarks.
The Danish translation is in, as well as the manual language switcher, let me know if that works for you. 
I really want to freeze the strings and get this release out, everything that's not terribly wrong can be fixed later!

FYr76, 2020-04-25:
Sure, you're right.  
Can't see the beta, maybe it takes a while.

FYr76, 2020-04-25:
#  corrections
1. I leave the minor language corrections for a later pull

# Language switcher

1. The languages begins with`auto`. I prefer `standard`
2. It's supposed to be `Español` 
3. it's supposed to be `Frysk` 

#  tutorial
1. Not important  for now, but maybe it's handy to relocate the `tutorial` to the main menu. It sound more logical to find it there, also because the tutorial starts with `Welcome` 

-------

### screenshot 1 / Frisian
[image]
1. I synchronised 1 sensor a few minutes ago. 2 sensors a few hours ago. They show Dutch text  `Gesynchroniseerd`  instead of Frisian

### screenshot 2 / Frisian
[image]
1. When pressing the graph, it shows date and time... problem is that the `names of the days`, the `months`  and the `abbreviations of the months`  are in Dutch
2. `uS/cm` -  `cm`  in Frisian is `sm` 

### screenshot 3 / Frisian
[image]
The same problem with the `abbreviations of the months`. They are in Dutch

-------

Now I'm changing languages to Danish...

### screenshot 4 / Danish
[image]
The same problem as mentioned above.... Dutch text, mixed with Frisian, but now in a Danish setting

### screenshot 1 / Frisian
[image]
You see what the problem is.... it still uses the devices language instead of the switched language and mixes  it up with

emericg, 2020-04-26:
> Can't see the beta, maybe it takes a while.

Yes indeed every text and software posted on Google or Apple store is subject to validation, and it can take from a couple of minutes to a couple of days :/

I changed espanol and frisk. Auto means automatic selection, it's pretty standard (haha) in this case, while standard language doesn't mean much.

Some strings can change language automatically while other needs either a reboot the app (so it's really an acceptable trade off) or an action (a sensor refresh will change most text and so update their translations) to act on the language change. This can't be fixed without reworking stuff, and reworking stuff means new bugs, etc...

This version was only supposed to have couples of bugfixes onto the last release done last month, then got new languages, then now has a whole lot of new code, so that's why I really want it out and new code in another release.

I can't thank you enough for your work on the new translations. Doing the translations on a mobile device really tells a lot about your motivation, and I believe we've been doing some real improvements on English too.
Doing them manually has proven to be a lot of extra work though, especially simultaneously changing source (English) strings and doing incremental changes on many translations files.
Every changes on source strings means updating translations for every languages, with the possibility to swap latest translation with older version of the (more or less) same string, especially when I don't speak the target language. Coupled with the many manual fixes on files, this has become really tricky.

So this release will be published like this, and then take your time while using the application to note where changes should be made.
If you want to discuss the source strings and meaning of the different text let's do it separately. It should probably come first. Things like untranslated units (like uS/cm) means adding them to the translated strings list, and so will have repercussions for every translation files. These files also need to be up to date with latest changes to the code or you risk translating already deprecated strings. 
If you want to improve translations, let's do it all at once. This is probably best done before the next release, to take into account every changes that will be made between now and then.
Like before, delete your fork, recreate it with last version of the source code, update every translations and push the changes.

FYr76, 2020-04-26:
Tnx, I understand how you see things. Complicated - I only see it from the end-user perspective - don't know anything about coding. It must be a battle on many fronts.
 It's great how you respond to input and ideas.  That really motivates me. 

I understood you prefer discussing the source text. I'm going to collect  ideas, bugs, etc the next weeks... and before the next update looking at the translations.

FYr76, 2020-04-28:
Busy with the Playstore text. What do you mean with Supports clock update? Supports clock update and changing temperature unit. 

You don't mention: North "Plant Monitor", and Vegtrug "Grow Care Home"`. I see Vegtrug is a growing brand in The NL. Same product, but maybe not clear to everyone.

emericg, 2020-04-29:
> It's great how you respond to input and ideas. That really motivates me.

Well I try to listen to inputs, but it's always a balance between the time I have, the time it would take to implement a feature, how many people would need and use it...

> I understood you prefer discussing the source text. I'm going to collect ideas, bugs, etc the next weeks... and before the next update looking at the translations.

I'll let you know when I'm close to a release. But in between I'm going to work on other softwares for a while...

> Busy with the Playstore text. What do you mean with Supports clock update? Supports clock update and changing temperature unit.

The device is a clock + thermometer, so the app can update the actual clock time and synchronize it with the time from the smartphone (I don't know if I say that right). It can also change the thermometer unit between Celcius and Fahrenheit.

> You don't mention: North "Plant Monitor" , and Vegtrug "Grow Care Home" . I see Vegtrug is a growing brand in The NL. Same product, but maybe not clear to everyone.

I knew about the Vegtrug "Grow Care Home" but I never saw the North "Plant Monitor" before. I believe both are just a rebrand of the Flower Care yes, BUT I never tested any of them so I'm a bit hesitant to advertise support for these devices unfortunately.

FYr76, 2020-04-29:
Okay, I'll finish the Dutch text
Looking to Playstore images of North and Vegtrug it's the same app as the official one. But indeed, you can't be for sure it's the same device, however  I have a Vegtrug and Xiaomi devices and they're exactly the same... On Aliexpress  I read it's a global rebranding

emericg, 2020-04-29:
So the Vegtrug works? Another user was apparently suggesting taht, but disappeared before I could get a confirmation.
Even if they renamed the device "advertised" name over bluetooth that would make them incompatible with the application. It would not be a hard bug to fix, but without a device... Still quite hard ^^

FYr76, 2020-04-29:
Yeah, it works exactly the same... only with newer firmware. I ordered through this link.

https://s.click.aliexpress.com/e/_B1tEly

-------

I finished translating, but first I looked  at the English text. Text in red are adjustments of the text found on https://play.google.com/store/apps/details?id=com.emeric.watchflower&hl=en_US

I chose  the word 'App' over 'Application' in both the Dutch and the Danish version. It's the most commen used word. if you want uniformity with 'Application', just say so, and I'll change it.

-------

# English

A plant monotoring application for Xiaomi "Flower Care" en "RoPot" sensors.

WatchFlower is a plant monitoring application that reads and plots data from your Xiaomi "Flower Care" and "RoPot" Bluetooth sensors. 

WatchFlower also works great with a couple of Xiaomi Bluetooth thermometers!

WatchFlower `is open` source software. By buying it here, you support its development!`

WatchFlower doesn't` track `your application usage`, doesn't `require account` creation, your GPS location, nor any other `personal data!`

The application is available in Danish, Dutch, Frisian, English, French, German, Spanish and (partial) Russian.

Application developed by Emeric Grange. 

Visual design by Chris Díaz.

Supported devices:

✔ Xiaomi "Flower Care"

✔ Xiaomi "RoPot"

- Works `with both international` and Chinese versions.

✔ Xiaomi "Digital Hygrometer Clock"

- Supports `automatic time setting` and changing temperature unit.

✔ Xiaomi "BLE Temperature and Humidity sensor" (with `LCD display`)

✔ ClearGrass "Digital Bluetooth Thermometer and Hygrometer" (with `E-ink display`)

About Bluetooth Low Energy:
- `On some Android devices, bluetooth is of lesser quality... unfortunately there's little that can be done. It doesn't do miracles, but Watch Flower has some tricks to improve your Bluetooth connection. Stay close to your sensors and let WatchFlower do the rest!`
- If the application `unexpectedly` doesn't work on your device, just uninstall it to get a refund! (see https://support.google.com/googleplay/answer/2479637)
- While scanning for Bluetooth Low Energy devices, the application `will ask, for "location permission". This is an Android requirement (see https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html#behavior-hardware-id). WatchFlower itself `does NOT` use your location, for any purpose.

Follow development, post bug reports and suggest new features on the GitHub page (https://github.com/emericg/WatchFlower) or `contact by` email (emeric.grange@gmail.com).
WatchFlower is NOT affiliated with Xiaomi (or any other sensor manufacturer).

-------

# Dutch

Een plantobservatieapp voor Xiaomi "Flower Care" en "RoPot" sensoren.

WatchFlower is een app om je plant te observeren. Het leest data van je Xiaomi "Flower Care" en "RoPot" bluetoothsensoren en toont dit in overzichtelijke diagrammen. 

WatchFlower werkt ook goed met een aantal Xiaomi bluetooth-thermometers!

WatchFlower is open source software. Door de app hier te kopen steun je verdere ontwikkelling! 

WatchFlower houdt je appgebruik niet bij, een account aanmaken is niet nodig, evenmin als je GPS-locatie of andere persoonsgegevens.

De app is beschikbaar in het Deens, Duits, Engels, Fries, Frans, Nederlands, Spaans en deels in het Russisch.

[... 46 lines cut ...]
Grafisk formgivnig af Chris Díaz.

Understøttede enheder:

✔ Xiaomi "Flower Care"

✔ Xiaomi "RoPot"

- Velegnet til både internationale og kinesiske versioner.

✔ Xiaomi "Digital Hygrometer Ur"

- Understøtter automatisk tidsindstilling og justering af temperaturenheden.

✔ Xiaomi "BLE Temperatur- og Luftfugtighedssensor" (med LCD-skærm)

✔ ClearGrass "Digital bluetooth-termometer og -Hygrometer" (med E-ink skærm)

Om Bluetooth Low Energy:
- På nogle Android-enheder er Bluetooth af mindre kvalitet ... desværre kan der gøres lidt ved det. Forvent ingen mirakler, men WatchFlower har nogle tricks til at forbedre din Bluetooth-forbindelse. Hold dig tæt på dine sensorer, og lad WatchFlower gøre resten!
- Hvis appen ikke fungerer på din enhed, skal du afinstallere den og få refusion! (se https://support.google.com/googleplay/answer/2479637)
- Scanning til Bluetooth Low Energy sensorer kræver "placeringstilladelse." Dette er et Android-krav (se https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html#behavior-hardware-id). WatchFlower hverken bruger eller gemmer din placering.

Følg appudviklingen, rapporter fejl og foreslå nye funktioner på GitHub-siden (https://github.com/emericg/WatchFlower) eller direkte via e-mail (emeric.grange@gmail.com).
WatchFlower er IKKE tilknyttet til Xiaomi (eller enhver anden sensor-producent).

emericg, 2020-05-13:
Allright thanks for the translations! I finally updated the store pages, and I checked the english translation with a native speaker to tidy up some sentences.

Time to put this issue to rest I think ^^ Thanks again for everything you've done, and I'll update you before the next version if you wish to make some modifications!

closed 2020-05-13 by emericg

emericg, 2020-09-15:
I'll be doing a release very soon (already in beta), if you want to have a look at some of the new strings.
There is not that many changes so it's only if you have some time.

Also, it would be really really really great if you could charge the files in https://github.com/lelegard/qtlinguist-installers/releases that would make both our jobs way easier, I cannot stress this enough.
You can translate the 3 langages at once, and all of the duplicated / deprecated / already translated strings are handled for you.

[14 more replies]