emericg/WatchFlower #12: Getting historic data from mi_flower sensors

Megachip, 2020-04-14:
`mi_flower` sensors storing historic values (so they do not need to pulled all the time). Not sure, if this is already reverse engineered.  
Maybe [this](https://github.com/sergem155/miflora-influxdb) could help.

emericg, 2020-08-12:
This is being worked on.
But the FlowerCare sensors force disconnect devices after a couple of seconds unless a crypto key is sent, I still need to figure that out...

DoCheckor, 2020-12-22:
Was just about to drop you an email if the software is really supposed to work like this but found the answer here. Running the desktop software and decided to leave the computer running at nights but it keeps crashing at some point of the night so it won't be a decent workaround. Anyway: Count me in for being highly interested on getting historical data synced... :-) Thanks!

emericg, 2021-01-17:
Mostly done \o/

[image]

emericg, 2021-04-25:
Anyone tried the feature on android? I haven't heard anything yet.

emericg, 2021-10-06:
By the way this is definitly done.

closed 2021-10-06 by emericg

lesleyblankendal, 2022-10-04:
History data doesn't seem to be syncing. It tries to connect but silently fails or doesn't get any data back. I see the history data when I check the Mi Flower Care app...

https://user-images.githubusercontent.com/4315309/193947879-a090072e-db0c-4e29-a2c1-a0f160b03378.MP4