emericg/WatchFlower #14: Add noting function

FYr76, 2020-04-17:
I would like to have a noting function with the possibility to add photo's, like in the original app. It gives other options to track the growth in the long term. What do you think about the following repository? 
https://github.com/7LPdWcaW/GrowTracker-Android

emericg, 2020-08-12:
I'm not sure I understand the feature.
Like an history screen / timeline, where you can post timestamped photos and notes (text fields) ?

FYr76, 2020-08-13:
Yes, you're right, just like a plant diary/ activity log. I used it a lot in the original app. A reminder / schedule function could be handy too. This app is a prettier example: https://play.google.com/store/apps/details?id=us.crema.vera

emericg, 2022-03-02:
Better late than never, but I'm working on this feature ^^ See attached pictures. Let me know if that's what you had in mind.

[image]
[image]

h3rm, 2022-03-02:
> Better late than never, but I'm working on this feature ^^ See attached pictures. Let me know if that's what you had in mind.
> 
> [image] [image]

yeah, niceee! can't wait until you release it ✌️

emericg, 2022-06-28:
Alright so this is now release!

> idea: can you maybe also add a function that makes a screenshot of the monthly overview of each sensor/flower care of "DATA"

Not easily no. I'll add the possibility soon to do next/previous months soon though, so the history graphs are more useful.

closed 2022-06-28 by emericg