emericg/WatchFlower #16: [Feature Request] Data Sync, Data Download, and Historical Data Fetch

philupnow, 2020-05-03:
Hi,

I recently downloaded WatchFlower on desktop (macOS) and iPhone for the graphing feature. Overall, great app and very polished looking on both platforms. Thanks for making this app! After working with the app on both platforms, I have a few new feature requests:

1) Please add the ability to sync data between the mobile and desktop app. I mostly use my phone to collect data around the house and in the garden but like to view the data/graphs on the desktop app. 
2) Please add ability to download data. The Xiaomi/North Plant Monitor has a data download feature that exports the data as a csv file (example attached, converted to xlsx for github), which is great for importing the data into other graphing and data analysis apps.
3) Finally, the sensors seem to have some on board memory for storing data. The Xiaomi app is able to collect the data points between syncs with the sensors but WatchFlower only collects a single data point for each sync. This is also reflected in the csv file which shows the sensor collects a data point every hour. This api gives the UUIDs and handles to read/clear the historical data: https://github.com/vrachieru/xiaomi-flower-care-api
Please give WatchFlower the ability to read and plot the historical data to make the graphing feature even better: more data = more fun :) !

Thanks,
Phillip
[2020-05-02-15-HHCC.xlsx](https://github.com/emericg/WatchFlower/files/4571498/2020-05-02-15-HHCC.xlsx)

SylvainLemieux, 2020-05-25:
I backup up Phillip for his ideas, mainly points 1 and 3.

Thanks,
Sylvain

emericg, 2020-05-27:
Hi,

The first point is (technically) possible, you just have to rebuild the app and set a "real" database address instead of the embedded sqlite database. I never actually tested it because I don't have a "real" database running but I've been wanted to for some time. If it does work I'd probably let an option for the users to use their own database.
Now, do I intend to provide this kind of online/cloud service for the app? Definitely not, sorry.

The second point has also been on the todo list for a long time but I never actually got past the point of fixing a file format. Thanks for providing the csv, I can probably start by emulating this format.

Yes this is long overdue too, please discuss this issue here #12

minosrex, 2020-07-06:
I second all points made!

Specifically the point on syncing exporting data, not just to analyse in depth but also to open up new interaction possibilities. For example, if you could sync the data with, say, a Google Sheet, then you can use IFTTT to monitor values, send communications based on those values, engage other systems (one example: the Gardena watering system, or the Netatmo weather station -which I already have :-)). It would mean, I guess, that you can leverage all the power of the IFTTT ecosystem without having to connect to it directly.

Best regards,
Simon.

emericg, 2020-07-10:
Where you able to test the new export button?
It save the file in your phone inside a "watchflower" directory. Maybe an "open with" button next to the save button would be better for your use case?

I know nothing about IFTTT and I don't have any device, but if a working (albeit not optimal) solution can be achieve I'm up for it.

minosrex, 2020-07-18:
So I was now able to nicely order the plants (using simple numbers for the location and doing "sort by location"... but they still synchronise in the original order :-) So now first sensor number 12 in the list syncs, then number 2, then number 6... It's worse than before :-D :-D

On the data sync, IMHO the suggested solution doesn't provide much value. You would have to manually do a lot of steps every few weeks (save the file to the smartphone, locate it, move it to google drive, then add the data from it to some master spreadsheet to draw the graphs and evaluations and IFTTT commands from).

I found some sources on how to connect an app to google drive here and make use that to make a backup of data here:
- https://medium.com/glucosio-project/backup-restore-a-realm-database-on-google-drive-with-drive-api-c238515a5975

On google drive API: https://developers.google.com/drive/android/intro

As you know by now I'm not a programmer so I let you judge how easy this is :-) What I CAN tell you is that integration of Google Drive with IFTTT is super easy and very powerful. It uses a very simple logic everyone can follow: define a trigger starting from a list of possibilities (an example in my case: "I am connected to my WiFi network in the daytime"), add a response to it also based on a list of possibilities (in this case: set the smart thermostat to 21°C). The trigger for Watchflower can then be something like "a certain cell in the Google Sheet exceeds a certain value, e.g. the soil is too dry", then the response is to activate the Gardena sprinkler). You don't have to programme or enable anything to make these scenarios happen, just enable the syncing of the data to somewhere where IFTTT can access them.

Cheers,
Simon.

emericg, 2020-08-12:
> So I was now able to nicely order the plants (using simple numbers for the location and doing "sort by location"... but they still synchronise in the original order :-) So now first sensor number 12 in the list syncs, then number 2, then number 6... It's worse than before :-D :-D

It's the same order as before actually. I'll see what I can do about respecting the order. You can manually select them and update them in the order you like though.

> I found some sources on how to connect an app to google drive here and make use that to make a backup of data here:

The best I can do is a button open with and you could select the drive app, that would streamline the process. Other than that it is widely out of scope for WatchFlower...

Other than that regarding the original issue,

1/ is still kind of `won't fix`

2/ is mostly done

3/ can be tracked in #12

closed 2020-08-12 by emericg