emericg/WatchFlower #17: Range on certain parameters inadequate

FlowersByIrene, 2020-06-14:
For some reason fertility and sunlight had been limited to 1k and 10k in this app. This is not ideal, as some plants has an upper limit for fertility at 2k and sunlight as much as 65k. I have a few semi-shadow plants who want 25-30k of sunlight. Topping out as low as 10k makes it difficult to assess whether they have ideal light conditions. This limitation is not in the stock app, but I really wish I didn't have to use that because - well - it sucks. Please update the range of these parameters to cover more plants.

emericg, 2020-06-15:
Hi,
I've upped the conductivity limit to 2k.

I've added a quick way to switch between "indoor" and "outdoor" mode for the luminosity. You can just click on the scale below the slider to switch modes.
I'm not particularly happy with that solution but it should allow you to set the values you want.
I like the little scale for indoor luminosity, but we can't have it if the values for outside are up to 100k.
What do you think?

[image]

FlowersByIrene, 2020-06-17:
Well, yes, it is usually outdoor plants that will have that kind of luminosity. Or someone with insanely powerful ceiling lights :-P In my case it's roses, and they thrive between 3k and 65k according to our Chinese friends. Quite a stretch, and probably difficult to get both the granularity and the range in the same slider. What if the outdoor mode simply extend the max to 100k? By the way, according to Wikipedia 120k is the max luminosity for direct sunlight. Would it still be possible to to set the values in steps of 1k with that many stops? I think some outdoors plants have a really wide span of acceptable light.

Thanks for the prompt response! Nice to see this project alive. 

Here's a screenshot of the values for roses:
[image]

emericg, 2020-06-19:
> What if the outdoor mode simply extend the max to 100k

That's what happen now with the new version just released, if you click on the legend the range will switch form 0-10 to 0-100k. The granularity is 5k (otherwise the two handle can overlap easily if they are close), but you can set the low limit with 1k granularity then switch to the 100k mode.
I'd like to do a more formal outside mode, with other changes like maybe different temperature range, more explicit freeze warning, but I need to think about it more.

emericg, 2020-07-06:
The new **beta** released today has a more explicit switch to toggle a proper outside mode.
Right now it's only saved if you set limits outside of the "inside" range. (Next version will have that value properly saved, but as it changes the database format I will do it together with other database changes related to the device history readings)
It only changes temperature range (0-50 outside, 12-32 inside) and luminosity (0-100k ouside, 0-10k inside), if you have other ideas let me know!
I will close this issue soon but you'll still be able to post messages in the thread.

[image]

closed 2020-07-16 by emericg