emericg/WatchFlower #19: cosmetic gui remarks

Roughtrade, 2020-07-09:
First of all thanks for the great software. The chinese original is very talkative and works without an internet connection. The rights request is also a joke. Your software is super exemplary ! 

Unfortunately now a few remarks:
The curve of the light intensity is not optimal for scaling. The blue curve and the dots are noticeably thicker than the others. Unfortunately, in the German translation the labels of the measured values are cut off on the left side. ( Android Version 1.4 )
![Screenshot_2020-07-09-20-43-45](https://user-images.githubusercontent.com/18134832/87078585-2c3d1300-c225-11ea-8ec9-565ad1c2d766.png
 )
Regards Peter

emericg, 2020-07-10:
Hi, thanks for the kind words!

> The curve of the light intensity is not optimal for scaling

Yes that curve is always problematic, I'm not sure what to do about it.
It wasn't enabled in previous version because of the scaling issues, and the difficulty to show both the usual or regular low values that you can have inside (0-5k) and the immediate jumps to very high values if you have full sun or a lamp. I could maybe use a logarithmic scale but even with that we could see a lot of spikes with day/night cycles...
I've enabled it again in 0.14 due to popular demand and to gather feedback.

> The blue curve and the dots are noticeably thicker than the others. 

That was intended :p because water would be the principal data. But that can be modified.
Also what do you find about the dots? Right now there is an option but I'm contemplating removing them.

> Unfortunately, in the German translation the labels of the measured values are cut off on the left side. ( A>ndroid Version 1.4 )

Thanks for the hour/minutes translations, that's the result of a recent change, they where indeed missing.

The labels cut of on the left side have been reported a couple of times too, but I don't really want to make them bigger because they would take half of the screens, and even more on little screens. Labels aren't more important than the values!
These abbreviations have been suggested before, what do you think?

Bodenfeuchtigkeit -> Just "Feuchtigkeit" or "Feucht." or maybe "Bodenfeucht."
Temperatur -> "Temp."
Lichtintensität -> "Lichtint." or only "Licht"
Fruchtbarkeit -> "Fruchtb."

Roughtrade, 2020-07-10:
This would I say:

Lichtintensität -> Leuchtkraft
Bodenfeuchtigkeit -> Bodenfeuchte
Fruchtbarkeit -> Fertilität
Temperatur -> Temperatur

Roughtrade, 2020-07-10:
Can the gaps in the daily histogram be closed by interpolation?  The sensors continue to measure in the gaps.  Apparently the hourly automatic measurement on my mobile phone does not work. 

The original app seems to bridge these gaps by reading the history stored in the sensor.

The darker fields for the night hours, I suppose, should be adjustable. For example, when a plant is in the growth phase it needs 14-16 hours of light.

[image]

Stay healty and ceep calm

Peter

emericg, 2020-08-12:
> Can the gaps in the daily histogram be closed by interpolation? The sensors continue to measure in the gaps. Apparently the hourly automatic measurement on my mobile phone does not work.

It's really only for measurements, the other graph already interpolate values. The application doesn't work in the background (yet), so the hourly measurement apply to when the app is on screen. See #20.

> The original app seems to bridge these gaps by reading the history stored in the sensor.

You can follow #12 

> The darker fields for the night hours, I suppose, should be adjustable. For example, when a plant is in the growth phase it needs 14-16 hours of light.

Maybe someday ^^ Right now there isn't much "per plant" settings, maybe if we have a plant database that would be an idea.

emericg, 2020-10-01:
Hi, I'm closing this issue because some things already made it to the current release and others are in the todo list. Thanks!

closed 2020-10-01 by emericg