emericg/WatchFlower #20: Background notifications on iOS and Android

fnutt, 2020-08-05:
First of all, thanks for a wonderful app and thanks for pushing it to App Store.

**Feature request:**
Notifications on iOS (and Android for those using it) would be highly appreciated. This would allow me to know when to water the plants instead of having to check the app regularly. Preferably it would be possible to set certain limits on moisture, temperature, luminosity etc., e.g. when moisture is below X  -> send notification.

Roughtrade, 2020-08-05:
In my opinion this is not really possible with this hardware. Your phone would have to constantly send requests to the sensors in the background to be able to send a notification. BLE is very vulnerable to power saving mode, especially with IOS. This already didn't work with the Corona app from Germany.
For such a thing you might need a mediator like an ESP32, which can send requests to the sensors and would provide them via WLAN. 
stay healthy and ceep calm

bvalat, 2020-08-05:
I would like notification as well. Should be possible to plan updates at some hours of the day to  refresh data and send notification.

fnutt, 2020-08-10:
Once every 6 hours as an example?

emericg, 2020-08-12:
Background service is in the work for Android. I really don't know if I can do it for iOS.
It would be once or twice per day, not more. Enough to have notifications triggered, not to have realtime graphs. Doing so every hour would kill both smartphone and sensors battery. For pretty graph, we would have to rely on #12 

Also, if the sensors are out of range when the background service start, that trigger many kinds of problems that have to be though through.

tjaadvd, 2020-12-04:
If you add a feature for iOS Shortcuts app to fetch data in de the app, that shortcut can be triggered via automations in the Home app.

emericg, 2022-06-06:
Who want's to try Android background updates? Send me an email with your Google account and I'll add you to the testers!