emericg/WatchFlower #22: Moisture doesn’t recognize

ezdeham, 2021-01-19:
Hi,

I have a Parrot pot and I just downloaded your iOS app.
Everything’s work fine except he doesn’t recognize the moisture. (See screenshot)
[image]

Can you please help me?

Best regards,
Daniel

emericg, 2021-01-19:
Hi!
Unfortunately the Parrot Pot support is still very theoritical... I've been trying to get one for some time, with no success so far.
There are a couple of differences between the pots and the flower powers, and there is no informations online (noone ever did a third party support for these devices).

I'm gonna publish a new WatchFlower version (probably tomorrow) with changes for the Parrot Pot , let me know if the values are better.
In the mean time I'm still trying to buy one to test properly and get everything working (including the automatic watering).

ezdeham, 2021-01-19:
If you live in Netherlands, I have one extra and I can send it to you.
It has a problem that it’s not working on battery and I don’t know what is wrong with it

ezdeham, 2021-01-24:
I did the update, apparently it works better but the new problem is numbers and percentages don't make sense.
[image]

emericg, 2021-01-27:
Yeah that's actually worse...
Unfortunately I live in France. I'll keep looking for a second hand unit with an affordable price (some people seems to think parrot pot are collectibles, and worth a lot...)

ezdeham, 2021-01-27:
I found a lot of them in marketplace of facebook in France.
Please check it:
https://www.facebook.com/marketplace/item/766040890838475/

emericg, 2021-01-28:
I don't use Fb I hadn't thought about it. I sent a few messages.
So far same problem, either ludicrous prices or no shipping...

emericg, 2021-02-22:
Just to let you know, I've sent another round of message to a couple of sellers, I didn't forget about the parrot pot!

emericg, 2021-03-05:
Alright it costed me 32€, but this will be FIXED in the next release. I also figured out how to read the water tank level.
I'll **try** to research how to do a manual watering, and maybe automatic watering (set a low limit to trigger watering) but that's really a MAYBE.

ezdeham, 2021-03-11:
Thanks a lot!
I can't wait until next update 😁
I hope you also find a way to automatic/manual watering!
Then I'll buy more of this pot

emericg, 2021-03-17:
I think the manual watering trigger is OK but it needs an UI, I'll do that later.
The history synchronization for the parrot device is documented, I think it's possible to make it work too. Again, later :p

emericg, 2021-05-24:
I'm closing this issue, if you want to add some informations feel free to do so.

closed 2021-05-24 by emericg