emericg/WatchFlower #23: support for androidx

awesomeerictech, 2021-01-29:
Hello emericg i love the project but it would be better if you moved to androidx and abandon v4

implementation 'androidx.appcompat:appcompat:1.1.0'
 implementation 'androidx.core:core:1.1.0'
  implementation 'androidx.annotation:annotation:1.1.0'

awesomeerictech, 2021-01-29:
Ahaa i made it work 

in class QShareUtils change

import android.support.v4.content.FileProvider;
import android.support.v4.app.ShareCompat;

To

import androidx.core.content.FileProvider;
import androidx.core.app.ShareCompat;

Then in build.gradle add 

implementation 'androidx.appcompat:appcompat:1.1.0'
implementation 'androidx.core:core:1.1.0'
implementation 'androidx.annotation:annotation:1.1.0'

Finally in Android Manifest change

    ￼    <provider android:name="android.support.v4.content.FileProvider"
                  android:authorities="com.emeric.watchflower.fileprovider"
                  android:grantUriPermissions="true" android:exported="false">
    ￼        <meta-data android:name="android.support.FILE_PROVIDER_PATHS" android:resource="@xml/filepaths"/>
    ￼    </provider>

to

   
<provider android:name="androidx.core.content.FileProvider" android:authorities="com.emeric.watchflower.fileprovider" android:exported="false" android:grantUriPermissions="true">
<meta-data android:name="android.support.FILE_PROVIDER_PATHS" android:resource="@xml/filepaths" />

awesomeerictech, 2021-01-29:
<provider android:name="androidx.core.content.FileProvider" android:authorities="com.emeric.watchflower.fileprovider" android:exported="false" android:grantUriPermissions="true">
<meta-data android:name="android.support.FILE_PROVIDER_PATHS" android:resource="@xml/filepaths" />
</provider>

closed 2021-01-29 by awesomeerictech

emericg, 2021-01-29:
Hi.
This is very interesting, I had not much knowledge about this "new" appcompat from androidx. I'll try to use it in my builds and see if the actual feature is still working (I believe you only tested the compilation).
This part of the code is only used to send the CSV export file to another app (instead of saving it to the SD card). But I have other apps that are making extensive use of the FileProvider, so it might be very useful.

Regarding the actual problem, you're probably just not using the same Android SDK than me (25.3.1) right?

awesomeerictech, 2021-01-29:
Hi ,I normally use androidx as a replacement for app compat libraries
wherever I am calling Java Code from Qt, all your projects including mini
infos use V4, I have converted all of them to androidx and it's rather fast
modern and supported by Android . I normally use component based software
engineering so I only use Themed Engine from your project rather than
creating it from scratch, I am using build tool 30 NDK 21 and API Level 30
, Android.

emericg, 2021-01-29:
That's pretty cool, I'm not aware of anyone else building my projects on Android. The appcompat library seems to be in the SDK not the NDK, out of curiosity do you know what version you are using?
Also, same question with Qt. I'm using 5.15 everywhere except on Android where I use 5.13, where more recent versions are just not working the same way... (I have problem with Qt Creator detecting Qt version and Android package creation)

I'll give androidx a try tomorrow then!

awesomeerictech, 2021-01-29:
I am using Qt 5.15.2 and I will be moving to Qt 6 very very soon, please
try androidx it is the one being developed and supported by android as we
speak and future versions of android will only be supporting androidx ,
again I also found it necessary to create custom build.gradle in your
projects please consider that too, thanks much for your big projects.

emericg, 2021-01-31:
OK I've tested it with MiniVideo Infos and it seems to work exactly the same. I've also updated the manifest, added a build.gradle and other stuff to work with Qt 5.15.
I'll do the same for the next version of WatchFlower tomorrow.

However Qt 6 will not be supported any time soon for most of my projects. They have removed many things that I use, like Qt Bluetooth, Qt Multimedia and also QtGraphicalEffects (?).

awesomeerictech, 2021-02-01:
Hi Qt 6 is not yet mature , you know Qt 5.15 is LTS, I think 3 years since
the first release, so I don't know what Qt company thought but I am hoping
they will be supporting those features too after LTS, yeeeah Android is
working fine , I am also thinking about adding a service to watchflower for
background information , myself too most of my projects are utilizing Qt
5.15.2 Qt 5.15.0 has several bugs please move to 5.15.2 which has fixed
most of them, I will be checking on updated minivideo and
watchflower,thanks.

emericg, 2021-02-01:
> please move to 5.15.2

OK I've pushed the changes to WatchFlower, it should be allright now.

> you know Qt 5.15 is LTS

Unfortunately Qt 5.15 LTS support is already over, unless you pay a commercial license... That sucks hard.

> I am also thinking about adding a service to watchflower for background information

If you plan to work on that please let me know so we can sync up.
At the moment I want to : 
- Finish the device history synchronization first for WatchFlower version 1.0 (that's *mostly* done)
- Then move to fix Android notifications for recent Android version (NotificationManager::updateAndroidNotification() was able to send notifications, but I think it needs an overhaul to work with API 26+, related to that https://codereview.qt-project.org/c/qt/qtandroidextras/+/292936)
- Then work on an Android background service. I have no idea how that would work, but you can already start WatchFlower in "daemon" mode without UI, so I guess that's where I would start.

awesomeerictech, 2021-02-02:
Let me try to find the right way of doing it using Qt remote objects and
android services and where it is relevant