emericg/WatchFlower #24: On iOS there is no safe space left for home indicator at the screen bottom

giedrius-stanevicius, 2021-02-03:
Notice how at the bottom iOS bar overlaps the app buttons (PS: don't worry about the sensor data - I'm just playing with my Arduino): 
[image]

emericg, 2021-02-03:
Indeed I never set that up. I'll have a look at it one of these days.
Glad to see someone making a custom device ^^ If you are using the v1 (on the master branch) there are many new capabilities for custom sensors (but the database format is not finished yet...).

emericg, 2021-03-03:
I'm closing this because this should be fixed in the next release of the app, and with other improvements related to safe spaces.
Feel free to reopen if this is not good enough!

closed 2021-03-03 by emericg