emericg/WatchFlower #25: On iOS an ugly UUID is displayed when plant location is absent

giedrius-stanevicius, 2021-02-03:
[image]

emericg, 2021-02-03:
Yes that's on purpose. That's useful to identify which sensor is wich, especially if you have multiple devices (a computer and a phone, so you can match them)
With the next version I should be able to set the mac address instead of these UUID on apple devices, would that be more acceptable?

giedrius-stanevicius, 2021-02-03:
Oh, I see. I thought it is broken as it is quite unexpected that some kind of ID would share the same place as Location. The biggest problem with UUID I think is its length - as you can see it does not fit the field. A MAC address would probably be better here, especially as you could check it with other BLE tools.

emericg, 2021-02-03:
On iOS and macOS the mac addresses of the bluetooth devices you connect is hidden from you, instead you get that UUID. But at least for the flower care, they write the mac address in a BLE characteristic (because you need to use the actual MAC in some of the communication between an app and the sensors).
I guess I can at least present it better or remove it alltogether in mac/iOS. I'll think about it.

giedrius-stanevicius, 2021-02-08:
I am thinking, if the UUID is just an internal ID, only useful to distinguish items on the screen, maybe it could be shortened, showing only some part of it.

emericg, 2021-02-12:
Yes I believe you are right, shortening these UUID would still allow to distinguish between them.
I will test lower font size and shortened UUID, it would be an improvement.

Also, showing the sensor name instead of an UUID when a plant is set but not a location would probably be better right?

giedrius-stanevicius, 2021-02-12:
Well, I don't know about sensor names and their semantics, but if they are unique and they come from the sensors, I guess in general they could be more useful as UUIDs, which as I understand are visible only to an app.

emericg, 2021-03-03:
Ok so I tried shortening the UUIDs, it's not really pretty. So at least I've made the font size of the UUID field very little so it's less visually displeasing...

giedrius-stanevicius, 2021-03-04:
Not sure if you did this, but I think removing curly braces could make UUID look better too.

giedrius-stanevicius, 2021-03-04:
Oh, and just noticed - uniform letter casing also could look better

giedrius-stanevicius, 2021-03-04:
Aaaad while you are at it, there is also a layout problem in some cases:
[image]

emericg, 2021-03-17:
I just picked up on the fact that the UUID is actually unique, it cannot be used to identify one sensor when using a mac and and iphone for instance.
This actually causes a load of trouble to a couple of new feature I'm working on... Anyway...
So, I removed it when on ios or macos. That should conclude this issue ^^

closed 2021-03-17 by emericg

emericg, 2021-03-17:
By the way let me know if you want to be on the testflight for the new iOS release!