emericg/WatchFlower #26: Invalid device name: MHOC401

juvid, 2021-02-12:
Hi!
I can't connect one of the supported device because the device name is incorrect in the source code.
MMC Digital Hygrometer 2 (EInk) - MHOC401
The correct name is "MHO-C401".

emericg, 2021-02-12:
Thank you for that, I still haven't received my units to test for myself...
Did you test these changes yourself?
Otherwise what version/platform are you using so I can release a new build for you to test.

juvid, 2021-02-12:
Yes, I have built the app on Windows and it is working with the correct device name.
I'm using the Android version also.
Thanks.

emericg, 2021-03-03:
Ok this should be fixed in the next release.
Do you by any chance know the firmware version / format of the MHO-C401 device?

juvid, 2021-03-03:
Thanks.

Model Number: MHO-C401
Firmware Revision: 1.0.0_0010
Hardware Revision: V_14
Software Revision: 0010
Manufacturer Name: miaomiaoce.com

emericg, 2021-03-04:
Alright thank you very much!
I'm closing the issue, feel free to reopen if the next release doesn't deliver ^^

closed 2021-03-04 by emericg