emericg/WatchFlower #36: Support for Android 10 and 11 , scoped storage

awesomeerictech, 2021-03-17:
Please add permissions for scoped storage in android 10 and higher

emericg, 2021-04-15:
It would be great to have but I have no idea how this works, either on Android side, and even less in Qt side...
Because WatchFlower uses storage only to write the data export file, and because this will continue to work on the close future, this will be very low on my todo list.

closed 2021-04-15 by emericg

awesomeerictech, 2021-04-15:
This means you might not be able to access storage without scoped storage
from android 11 api or rather API LEVEL 30, that is too sad isn't it ?

emericg, 2021-04-15:
Right now I use API 30 + android:requestLegacyExternalStorage="true", and it works.
Even if it doesn't, instead of clicking "save file" you can just click "open file with" so there is no need for filesystem access.

awesomeerictech, 2021-04-15:
yeah it works i also use android:requestLegacyExternalStorage="true" in my
apps but future apps won't support this in manifest according to google
hence the need to upgrade to scoped storage, otherwise we can have that for
now