emericg/WatchFlower #40: [Feature Request] Flatpak release

AurelienKun, 2021-04-13:
Please, could you release Watch Flower on Flatpak ? 
It has so much better desktop integration than AppImage.
Thank you !

I love your app ! You've done an amazing work. 
*High Five*

emericg, 2021-04-15:
Hi and thanks for the kind words.
I'd like to do that, that being said FlatPak integration is nowhere near as straightforward as any other packaging system. It might take me some time. I'll poke around and see what I can do.

And you just need to have AppImageLauncher installed to have automatic integration of the AppImage files. That's actually convenient.

emericg, 2021-06-18:
Do you know your way around flatpak packages?
I've created a manifest here : https://github.com/emericg/WatchFlower/blob/master/assets/linux/watchflower.flatpak.yml
And set up a github action to build it every commit : https://github.com/emericg/WatchFlower/actions/runs/950300755

I'd like feedbacks before I even try to upload that to flathub.

AurelienKun, 2021-06-18:
I'm sorry I have no idea how to make a flatpak package ^^'

AurelienKun, 2021-10-31:
@emericg I saw the 3.0 release note and noticed the flatpak release. But I couldn't find watchflower on flathub. Where did you published the app ?

emericg, 2021-10-31:
Yes actually that meant the flatpak manifest should be ready to be submited to a flatpak "store" or whatever they call these.

I've just submited an entry to flathub, we'll see how that goes: https://github.com/flathub/flathub/pull/2610

emericg, 2021-11-09:
Allright the app is now available on flathub! https://flathub.org/apps/details/io.emeric.watchflower

AurelienKun, 2021-11-09:
Thank you ! I'm closing the request then !

closed 2021-11-09 by AurelienKun