emericg/WatchFlower #41: Sensor doesn't show graph dots

brechtneven, 2021-04-23:
First let me compliment you with your app, Very well made!

I just got some Xiaomi sensors and noticed that suddenly one of them isn't able to show graph dots anymore, which I find a very nice feature. All other behave like they should. Also I just noticed that battery level is at -1% and firmware is UNKN. 
Didn't had this issue yesterday. Since I just got them, I can still return them. You think I got a faulty sensor?

emericg, 2021-04-25:
Hi!
The graph dots are shown until there is 16 data records or more. Maybe I can set the limit a little higher. However with a synchronized sensor and hundreds of records like below it would be unreadable with points.
[image]

Also battery and firmware not set is not really important, and probably a bug in the app not a faulty sensor. I'll have a look I might know why it could happen.

emericg, 2021-05-24:
I'm closing this issue, if you want to add some informations feel free to do so.

closed 2021-05-24 by emericg