emericg/WatchFlower #42: [Feature request] Blacklist device(s)

AurelienKun, 2021-04-24:
Hi, I love this app. It is so great that I bought more sensors for my house. 
Unfortunately, the application detects also my neighbour's sensor as I live in apartment in a city. 
So, could it be possible to implement a blacklist somewhere to avoid this situation ?

Thank's a lot 👍🏻

emericg, 2021-04-25:
Hi.
Yes that sounds like an useful feature.
I thought about it in the past, using the wireless connection strength to limit the scanning "range". I'll try to think it through and come up with a proper feature.

In the mean time you can delete the unwanted sensors using long press on a sensor, or middle mouse click right?
Sensors are only added after a scan, so at least they should not reappear each time you use the app.

AurelienKun, 2021-04-25:
Oh nice ! I didn't know about the middle click (Linux version here). Good to know ! Thank's ! It's a good workaround for now ^^
Do I leave this ticket open until you implement a more proper way ? @emericg

emericg, 2021-04-27:
Sure you can leave this issue open, I'll update it when progress are made.

emericg, 2022-03-02:
I'm closing this issue, with the next version you will be able to blacklist any device around you:

[image]

closed 2022-03-02 by emericg