emericg/WatchFlower #45: Keeps crashing on windows 10

cyberb0b, 2021-05-09:
Im sorry to report It error here with event 1000 application error. 
Its a clean 1.0. Added all my devices (50) but after a few moments the application closes and there is an error 1000 in my event viewer. 

–––

Log Name:      Application
Source:        Application Error
Date:          5/9/2021 19:20:10
Event ID:      1000
Task Category: (100)
Level:         Error
User:          N/A
Computer:      WINDOWS10-VR
Description:
Faulting application name: WatchFlower.exe, version: 1.0.0.0, time stamp: 0x606743ed
Faulting module name: Qt5Bluetooth.dll, version: 5.15.2.0, time stamp: 0x5fa51110
Exception code: 0xc0000005
Fault offset: 0x0000000000043b6a
Faulting process id: 0x4cec
Faulting application start time: 0x01d744f77e064cc3
Faulting application path: C:\***\WatchFlower-1.0-win64\WatchFlower.exe
Faulting module path: C:\***\WatchFlower-1.0-win64\Qt5Bluetooth.dll
Report Id: 5d3ea902-15c1-46b1-85b7-39843d93933a
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
 <System>
   <Provider Name="Application Error" />
   <EventID Qualifiers="0">1000</EventID>
   <Version>0</Version>
   <Level>2</Level>
   <Task>100</Task>
   <Opcode>0</Opcode>
   <Keywords>0x80000000000000</Keywords>
   <TimeCreated SystemTime="2021-05-09T17:20:10.2679645Z" />
   <EventRecordID>398730</EventRecordID>
   <Correlation />
   <Execution ProcessID="0" ThreadID="0" />
   <Channel>Application</Channel>
   <Security />
 </System>
 <EventData>
   <Data>WatchFlower.exe</Data>
   <Data>1.0.0.0</Data>
   <Data>606743ed</Data>
   <Data>Qt5Bluetooth.dll</Data>
   <Data>5.15.2.0</Data>
   <Data>5fa51110</Data>
   <Data>c0000005</Data>
   <Data>0000000000043b6a</Data>
   <Data>4cec</Data>
   <Data>01d744f77e064cc3</Data>
   <Data>C:\***\WatchFlower-1.0-win64\WatchFlower.exe</Data>
   <Data>C:\***\WatchFlower-1.0-win64\Qt5Bluetooth.dll</Data>
   <Data>5d3ea902-15c1-46b1-85b7-39843d93933a</Data>
   <Data>
   </Data>
   <Data>
   </Data>
 </EventData>
</Event>

emericg, 2021-05-24:
Thanks for the detailed report guys.
So far I have never be able to reproduce that crash, but many people complained so I'll have another look at this soon.

> last edit :D ￼ it seems that wf 1.0 has qt 5.15 onboard. maybe an update to 6.1. works? :)

Qt 6 does not have any bluetooth support at all, so probably not ^^

seeyouchiu, 2021-06-19:
I have another issue with the Bluetooth connection between Wins 10 PC.
Any guys encounter PC can view parrot pot on the Bluetooth list but can't connect successfully.
Only my Ipad can success so far

janilahti, 2021-07-01:
This issue was originally reported for WatchFlower version 1.0, but just I found the software, already on version 2.0. I have a similar problem with Windows 10. WatchFlower may start nicely and work for an hour (until next BLE update), or may hang right at the start so that I need to restart it a few times until it just starts to work again. For Bluetooth I'm using whatever Lenovo put inside a Thinkpad and default drivers. Two MiFlora sticks with firmware 3.2.4.

h3rm, 2021-07-06:
@janilahti i posted the problem before, but it persists in v2.0 too... can you maybe check if you use a realtek bluetooth driver on your thinkpad? the past few weeks i read a few times that win 10 realtek-driver can cause problems with the automatic installed microsoft-provided realtek-driver... but i wasn't able to find a solution.. i tried a) the realtek bt onboard driver on my hp 250 g4 notebook and three different BT-usb-sticks 4.0 and 5.0... the only thing i found out: all of the bought usb-dongles have realtek chips inside;)

1. i tried the original hp 250 g4 realtek driver from hp directly, also checked this one with driverbooster -> still application crash
2. i tried to update the hp 250 g4 driver with "driverbooster" (i know it's not the best software/manufacturer, but it does what it shall do). the tool detects the correct chip and installs the newester driver -> still crash
3. i tried on all 3 BT-USB-sticks at first:
3.1 the driver provided by the sellers -> still crash
3.2. on all the sticks the same thing with driverbooster-driverupdate (before this -> clean-driver-deinstallation + restart) -> still crash
4. i also tried to check the hardware-id of each stick and tried to find an "official" driver (eg. USB\VID_0BDA&PID_8771&REV_0200
) -> searched in on google to find out the chip MPN/model -> should be RTL8671B -> no driver from realtek directly: https://www.realtek.com/en/component/zoo/advanced-search/57475?Itemid=276

@emericg wrote that there are more users with the same problem with qt 5.15 and bluetooth.. but i am too stupid to understand and find something in jira (https://bugreports.qt.io/browse/QTBUG-94001?jql=text%20~%20bluetooth%20ORDER%20BY%20created%20DESC) and don't know anything more about "QT" :)

i still hope @emericg finds a solution and maybe he can tell us what causes/caused the problem :)

h3rm, 2021-07-30:
@emericg https://wiki.qt.io/New_Features_in_Qt_6.2 juhu :) "The Qt Bluetooth module has been ported to Qt 6 and the CMake build system".. will this help?

ZeeHammar, 2022-05-13:
I am having the same issue with WatchFlower 3.1 constantly crashing. I am showing the same Qt5Bluetooth.dll error in the windows application event log reported by the OP (included below). I am not familiar with the QT framework, but doing some searching I think there are others who have reported similar issues with the bluetooth implementation and referenced potential workarounds. I could be way off the mark though. 

Log Name:      Application
Source:        Application Error
Date:          5/4/2022 3:25:22 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
Description:
Faulting application name: WatchFlower.exe, version: 3.1.0.0, time stamp: 0x618173ad
Faulting module name: Qt5Bluetooth.dll, version: 5.15.2.0, time stamp: 0x5fa51110
Exception code: 0xc0000005
Fault offset: 0x0000000000043b6a
Faulting process id: 0x43dc
Faulting application start time: 0x01d85fecae95a223
Faulting application path: C:\Program Files\WatchFlower\WatchFlower.exe
Faulting module path: C:\Program Files\WatchFlower\Qt5Bluetooth.dll
Report Id: a74e9a71-4451-41ec-8e57-57be5776c0ad
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-05-04T20:25:22.6192133Z" />
    <EventRecordID>102423</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Security />
  </System>
  <EventData>
    <Data>WatchFlower.exe</Data>
    <Data>3.1.0.0</Data>
    <Data>618173ad</Data>
    <Data>Qt5Bluetooth.dll</Data>
    <Data>5.15.2.0</Data>
    <Data>5fa51110</Data>
    <Data>c0000005</Data>
    <Data>0000000000043b6a</Data>
    <Data>43dc</Data>
    <Data>01d85fecae95a223</Data>
    <Data>C:\Program Files\WatchFlower\WatchFlower.exe</Data>
    <Data>C:\Program Files\WatchFlower\Qt5Bluetooth.dll</Data>
    <Data>a74e9a71-4451-41ec-8e57-57be5776c0ad</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

emericg, 2022-06-07:
Can you guys give the 4.0 version a go?

You can download it from the CI builders:
https://github.com/emericg/WatchFlower/actions/runs/2456127012

ZeeHammar, 2022-06-09:
Haven't had any issues with crashing on version 4.0 yet. I will leave it running for a few days to verify.

ZeeHammar, 2022-06-20:
Issue still seems to be present, but it occurs with much less frequency. I can't seem to make it 24 hours without a crash, but before it was crashing every other sync attempt. 

**Error Details:** 

Faulting application name: WatchFlower.exe, version: 4.0.0.0, time stamp: 0x629f8403
Faulting module name: Qt6Bluetooth.dll, version: 6.3.0.0, time stamp: 0x624cbbd4
Exception code: 0xc0000005
Fault offset: 0x000000000005c795
Faulting process id: 0x4fe4
Faulting application start time: 0x01d87c65bbb105b5
Faulting application path: C:\Program Files\WatchFlower\WatchFlower.exe
Faulting module path: C:\Program Files\WatchFlower\Qt6Bluetooth.dll
Report Id: c65c9279-5157-4ea6-9bac-cb3e59b2c406
Faulting package full name: 
Faulting package-relative application ID:

emericg, 2022-06-20:
Well I guess that's progress :)
Thanks for testing!

cyberb0b, 2022-06-20:
Here it works fine now.

emericg, 2022-07-22:
I'm closing this thread, as the "keep crashing" parts seems fixed at least. I'll keep an eye on the "crash after 24h hours eventually", as it as been reported a couple of times.

closed 2022-07-22 by emericg