emericg/WatchFlower #47: WatchFlower.exe just hangs with a blank screen

Stan-Gobien, 2021-09-17:
**Describe the bug**
The program worked fine for a couple of weeks.
Now since a couple of days it won't open anymore. The program starts to the Windows tray, but on right click "show" or "settings" a blank screen open with the title "Watchflower (not responding)" and just hangs. Waited a couple of minutes but no success. Taskmanager shows WatchFlower.exe with a constant 6% CPU (16T CPU, shows 1 thread almost at max constantly) and a steady 85.224KB RAM with no change. Seems like it's stuck in some kind of loop.

**Expected behavior**
I expected the program to open.
It worked (sort of) fine for a couple of weeks.

**To Reproduce**
Don't know how to reproduce on your system

**Screenshots**
[image]
[image]
[image]

**Your environment**
Windows 10 x64 21H1
AMD Ryzen 7 3700X / 32GB RAM / SSD storage
Watchflower 2.1 x64 for windows zip from the site

**Additional context**
I tried renaming the "DB" file in %AppData%\WatchFlower and a new database is created but the problem is not solved.

Stan-Gobien, 2021-09-17:
And ofcourse, after submitting the bug report I decide to check if WatchFlower saves settings in the Windows Registry.
Computer\HKEY_CURRENT_USER\SOFTWARE\WatchFlower
Deleted the Key, and sure enough the program starts again.

So problem solved (for now).

Stan-Gobien, 2021-09-18:
Problem has reappeared now (the next day) when I start up my computer.

Deleted the reg key again, and works again.

emericg, 2021-09-20:
Hi.
This is a bit weird given nothing has changed on WatchFlower side, but why not, with Windows you never know...

Can you try the newest version first: https://ci.appveyor.com/project/emericg/watchflower
There are some changes on the code that handle multiple launches (but maybe it's worse now :p)

Otherwise, to try and pinpoint what's wrong, can you try deleting only a couple of register keys from the WatchFlower folder?
All of the keys from ApplicationWindow/x,y, width, hieght, visibility... can have an impact on startup. Maybe settings/startMinimized and settings/trayEnabled). The others, very unlikely ...

emericg, 2021-09-20:
A more direct link to download the latest version: https://ci.appveyor.com/project/emericg/watchflower/build/artifacts

Stan-Gobien, 2021-09-20:
It was working today without issues.
Nevertheless, I updated to the latest build using your link.

emericg, 2021-09-21:
Alright keep me posted if anything else happens.

emericg, 2021-10-06:
Please re-open this issue if you have any new informations.

closed 2021-10-06 by emericg

Stan-Gobien, 2021-12-02:
Happens again.
[image]

Watchflower remember the last position, on my secondary screen but hangs when I open it by double clicking on the tray icon (autostart).

Stan-Gobien, 2021-12-02:
I looked at the registry key "HKEY_CURRENT_USER\SOFTWARE\WatchFlower\WatchFlower\ApplicationWindow"
The value for x was fffff9ee
The value for y was ffffffff

I set both values to 0 and now Watchflower opens fine on my primary screen.

I dragged the window back to the approximate same location on my secondary screen and closed the window.
The registry hex values for x and y are updated with new values (x = fffff66b; y = a7).
I closed the window and try to reopen by double click, it hangs again.

emericg, 2021-12-05:
Ok thanks for the detailed feedback, that is very valuable information!
I'll see if I can test on a second monitor, but I can already check the code for a few things related to that now.

I'll let you know what I find.

Stan-Gobien, 2021-12-05:
So far it's working fine on my primary screen.