emericg/WatchFlower #50: Issues on Android platform, device history  [Android 11]

TUstudents, 2021-09-29:
Hi,

I think I am one of the few people who installed it on a mobile device. I was curious, if I can exchange the proprietary FlowerCare App with GNU software. In overall a lot of effort has been put in WatchFlower for a usable app, that I want to appreciate.  However, some things do not work as Intendant or are missing, but first I want to give feedback of core functionalities.

**Bug**
Registered devices are synced on app start, but syncing with device history needs a manual start for every device with the "Update history" command. The sync routine will not finish, blocking other threads of the app until it is force closed. However, synced history data is visible in the app.

**Expected behavior**
After history syncing of one device the "update history" should be available for other devices again. Ideally, history syncing should be made available as part of the auto sync queue routine for every device

**Details**
After multiple scanning of sensors I was able to register 2 MiFlora produced a few years ago, 2 VegTrugs and 2 Miflora of the newest generation produced June 2021. Additionally 1 Thermometer was registered and one themometer with alternative firmware removed. Thermometer history is not supported in WatchFlower.
After every history sync the app will no longer follow new commands like update, blink led,.. but history values are displayed in app. History is not deleted from device after sync, which is very good as history can also be imported to FlowerCare, which resets history after import in Mi's database. I think the sync routine just never finishes or is throwing an error.
In logcat, errors are logged from 
>BufferQueueProducer
>[SurfaceView - com.emeric.watchlfower/
>com.emeric.watchflower.NotificationDispatcher#1]
>(id:3c3000027b2, api :0, p: -1, c963) dequeueBuffer:
>BufferQueue has no connected producer

But from a quick google search this is unreleated to bluetooth and an issue with flutter or camera. 

**Environment*
Android 11 on a rooted device, custom firmware similar to Lineage OS

emericg, 2021-10-04:
> I think I am one of the few people who installed it on a mobile device. I was curious, if I can exchange the proprietary FlowerCare App with GNU software. In overall a lot of effort has been put in [...]

If you build it yourself on Android then yes I think you might be the first!
The version published on the store is using the exact same code than the open source version, but:
- the package signature is different
- I deploy a patched version of Qt for better Bluetooth advertisement support, like its mentionned in the readme (https://github.com/emericg/WatchFlower#using-flowercare-live-mode-and-historical-data). These patchs will be included in Qt 6.3 (probably).

> The sync routine will not finish, blocking other threads of the app until it is force closed

This is weird, it's obviously not supposed to do that. If you are running your own version you should be able to use the debugger to check what's blocking the app.
The history sync can be run while the app is refreshing other sensors, and is very tolerant to the bluetooth communication being stopped without warning (happens all the time...).

> one themometer with alternative firmware removed

Yeah I don't support those.
I write documentations for device APIs (like this https://github.com/emericg/WatchFlower/blob/master/docs/thermobeacon-ble-api.md), if these custom firmware have a common and documented API it might be worth doing it. Otherwise it's really not worth the efforts.

> Thermometer history is not supported in WatchFlower.

Some have history support (it's mentionned on the website: https://emeric.io/WatchFlower/#devices). But every device needs a proper reverse enginering of its API, and this takes time an efforts...

emericg, 2021-10-07:
> Hi, no, I did not compile the app on my own, I am using the playstore app v2.1 from 8.7.2021

Then no, you are only one of the few thousands to installed it on a mobile device ^^

> logcat logs multiple errors on the GUI interface (SurfaceView)

The graphics related errors from the logcat are irrelevant, they happen all of the time with that kind of app.

The problem you have is either related to the state of one device getting stuck somewhere (but I can't say I saw that before though) or just a bug in the menu (so you can probably still trigger actions from the main device list, by long pressing on a sensor).
When everything is disabled exept "clear history", I'm pretty sure there is no actual way to do that... So I guess it is a bug of the menu?
When you start syncing history for instance, the actions are not disabled, if you click them again it just won't do anything.

The code related to device state has changed a lot since the version 2.1 (for the best I hope), so I guess you'll have to check again against version 3, that I plan to release soon (it will have fixes for your other problem too).
In the mean time I'll try to figure out if something can get the three dots menu stuck, we never know...

emericg, 2021-10-11:
A beta release on Android should be available (very soon). There should be improvements for the bugs you reported.
If you can give it a try that would be much appreciated.

emericg, 2021-12-14:
I found out why the buttons were not behaving like they should, a stupid mistake on my end. This will be fixed with the incoming beta release.
That should conclude this issue?

closed 2022-01-07 by emericg

TUstudents, 2022-01-10:
Is the V3.2 with this fix coming to the Google store? No update on the beta channel so I moved to main and there is no update either.

emericg, 2022-01-10:
I don't understand, the 3.2 is supposed to be in the Android Beta channel, it doesn't appear there?
https://play.google.com/store/apps/details?id=com.emeric.watchflower

TUstudents, 2022-01-11:
I applied to the beta channel again. Now it updates to 3.2.
I can confirm GUI button update bug has been fixed and is working.

emericg, 2022-01-11:
Alright good news!