emericg/WatchFlower #51: Device database initialization error

TUstudents, 2021-09-30:
**Describe the bug**
If a new device is added (maybe only the first in the database), and the history is synced the first database entry is initialized with wrong values. The first entry in the history is Moisture 255%, Temperature -0.1°, Luminosity -1 lux, Fertility -1µs/cm (which is probably -1 cast into uint 8bit, -1 for floats as temperature is reported in 0.1°C units AFAIK. These artifacts are present both on the Windows and Android app and probably for all platforms. While not a serious issue itself, this bug is affecting the auto-scaling, as 255% moisture can not be reached in life operation and all subsequent values are scaled down to tiny bars.

**Expected behavior**
Initialize moisture to 0% and do not use first value for auto-scaling at all

**Screenshots**
[image]

**Your environment**
Win10 WatchFlower v2.1
Android11

emericg, 2021-10-04:
I *think* the FlowerCare is returning wrong values, as you mentionned probably -1 on all values.
My default values in the code are actually -99 ^^ (because -1 is a valid value in many cases).

Now that being said, I never saw this particular case, but I can set the app to not register values when we get -1 on all sensors. 
I'll add that for the next release!

emericg, 2021-10-06:
This [commit](https://github.com/emericg/WatchFlower/commit/4c8303c38b607478e3170827f3c14efcccedd2cd) should fix the issue you mention (and more similar it) by preventing these values to ever make it to the database.
Can you try latest version and tell me if it works for you?