emericg/WatchFlower #54: Watchflower pasted from the clipboard

fluxens, 2021-12-14:
**Describe the bug**
Got a message telling  me that watchflower accessed my clipboard.

**Expected behavior**
Should not access my clipboard. what's happening here?

**Screenshots**
[image]

**Your environment**
Android 12

emericg, 2021-12-14:
Another breaking change from Android 12, add it to the list...
WatchFlower doesn't use the clipboard (at all), but somehow it's producing a warning anyway.

It's probably a bug (?) in a software library that is used by the app, it will be tracked here with other developers having the same problem: https://bugreports.qt.io/browse/QTBUG-98412

emericg, 2022-03-10:
I'e updated the app to the latest Qt 6 release, this bug should be fixed. I'll let you know when the beta is available for testing.

closed 2022-06-24 by emericg