emericg/WatchFlower #55: No sensors detected on Windows 11

coolmyll, 2022-01-09:
I bought a usb bluetooth adapter for my pc but the app is not detecting any of my sensors. The adapters seems to be installed correctly (appears on the taskbar on windows) but the app is not finding the sensors.
The app works and find the sensors correctly both on my phone and on my tablet though.

**Expected behavior**
To find the sensors

**To Reproduce**
Not sure how to reproduce in another pc. Does the bluetooth adapter need to be internal? Does the app not work with usb adapters? Are there any error logs i can check?

**Your environment**
Windows 11 PC with a Bluetooth 5 usb adapter.

emericg, 2022-01-12:
I'm sorry, not much I can do.
I never tried Windows 11, maybe there is a little something missing in order for this to work. Also it may simply be an unsupported adapter. Can you tell us more about the one you have?

> Does the bluetooth adapter need to be internal? 
> Does the app not work with usb adapters?

Any Bluetooth card should work, internal or not

> Are there any error logs i can check?

Sorry there isn't

coolmyll, 2022-01-18:
Its a generic usb one (but seems to be working / detected on windows).
I dont think that windows 11 is the problem here. Seems unrelated.

I want to help but i am not sure how/if i can help here. If there are no logs we dont know if the bluetooth is not detected or if something else is wrong.

emericg, 2022-01-18:
If no bluetooth adapter is detected, you should see a banner like this one:

[image]

coolmyll, 2022-01-27:
Sorry for the late reply. No i dont have this message. It looks like this and doesnt detect anything.
[image]

NOTE: It looks like this even if i dont have the adapter connected. And everytime i start it i get the intro cards. So something else must be wrong.

h3rm, 2022-03-08:
> 

do you find any "flower care" sensors if you manually search for bluetooth devices in windows directly (normally you find all sensors there - but you can't pair them)? if not maybe there is a driver problem (maybe completely uninstall the bluetooth driver+ bluetooth radio driver via device manager.. sometimes the drivers are broken if you used other BT adapters before) and/or BT LE isn't really supported. 

Try eg. LogiLink BT0015 (Bluetooth 4.0 (LE, Klasse 1, EDR, DualMode)) or https://aliexpress.com/item/4000414075718.html

I don't have 1.3 installed but 1.0 finds the sensors with https://aliexpress.com/item/4000414075718.html without problems on windows 11

emericg, 2022-03-10:
Can you try the new builds from here : https://github.com/emericg/WatchFlower/actions/runs/1958745924
It's very much in development but it got many fundamental changes on Bluetooth and Windows support. Also there is a "device detection scanner" so you can check if any device appear using that feature :

[image]

coolmyll, 2022-03-10:
@h3rm 
[image]
I think some of them are the sensors.
@emericg 
[image]
I guess I can order one of the above but I wanted to make sure its not a bug or something else. To be honest my 2 sensors is in the next room so behind wall (works with my android though) but the xiaomi temperature sensor is very close to the bluetooth adapter.

emericg, 2022-03-10:
Yeah, it definitely looks like a bug and not a problem with your BLE adapter...

h3rm, 2022-03-10:
> Can you try the new builds from here : https://github.com/emericg/WatchFlower/actions/runs/1958745924 It's very much in development but it got many fundamental changes on Bluetooth and Windows [...]

wow, very nice "killer"-feature.. good job! maybe you can also show the battery status icons for each sensor in the device map view? would be cool :) everything else works fine on linux (ubuntu) but when i sync history for all devices the app crashes at the moment (i can reproduce that)

> @h3rm [image] I think some of them are the sensors. @emericg [image] I guess I can order one of the above but I wanted to make sure its not a bug or something else. To be honest my 2 sensors is in [...]

as emeric already wrote: if the flower cares can be found directly via windows bluetooth search its not a driver or adapter problem normally

emericg, 2022-03-10:
@coolmyll At the moment I don't really know where this is coming from. I've rebooted on Windows to check if everything was still working on my machine and it is...

@h3rm Well wait you see the plant database and the android notifications :)
Regarding the journal you can leave comments in the other thread with mockup or anything, I'm interested in getting feedback on the feature.
(Also the entries are not persistent yet, they reset every time you open the app)

rngarcia, 2022-05-13:
Guys, this seem to be a Windows 11 problem with BLE devices... Saw this here: https://answers.microsoft.com/en-us/windows/forum/all/bluetooth-ble-doesnt-work-properly-on-windows-11/1bedd958-45c7-453f-9e24-f62e41bca851?auth=1 and I'm also having trouble adding the Flower Care sensors on my machine. WatchFlower won't find them, but neither will Win 11.

emericg, 2022-07-22:
You can try again with the [latest builds](https://github.com/emericg/WatchFlower/actions/runs/2616589731) or the new releases, that have many improvements related to this issue.

I'm closing this thread. Feel free to reopen if the problem still occurs.