emericg/WatchFlower #56: Android BUILD: error: resource android:attr/lStar not found

garudaonekh, 2022-01-12:
Hi,
I got this error when I build on Android.
`error: resource android:attr/lStar not found`

It looks like Android X issue. But can't find where to fix it.

Thanks;

emericg, 2022-01-12:
Can you edit the assets/android/build.gradle file and try with this dependency section?

```
dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar', '*.aar'])
    implementation 'androidx.appcompat:appcompat:1.3.0'
    implementation 'androidx.core:core:1.3.0'
    implementation 'androidx.core:core-ktx:1.6.0'
}
```

garudaonekh, 2022-01-12:
Thanks, That solved the above issue. However, I got another error with `android:usesPermissionFlags `. I use Android 12(31) however, still issue. So I remove this android:usesPermissionFlags from the manifest and it works.

closed 2022-01-12 by garudaonekh

emericg, 2022-01-18:
Did you removed just the android:usesPermissionFlags="neverForLocation" or both the BLUETOOTH_SCAN and BLUETOOTH_CONNECT permissions altogether?

garudaonekh, 2022-01-19:
> Did you removed just the android:usesPermissionFlags="neverForLocation" or both the BLUETOOTH_SCAN and BLUETOOTH_CONNECT permissions altogether?

Just the useFlagpermission