emericg/WatchFlower #57: Grow Care Garden(Flora Care MAX) issue

garudaonekh, 2022-01-12:
Hi,
There're several glitches with my Grow Care Garden sensor.( VegTrug "Grow Care Garden" (GCLS002). Firmware: 3.6.6

1. Fertility and Moisture disappeared. I kill the app and restart again and it appeared.(in the image)
[image]

2. Blink LED function not work. It worked with Mi Flora(Indoor version)
3. History data is wrong. The temperature displayed is 31.6 but in history it showed 9.6 celcius. Our country is hot and never go below 20. Fertility also the same issue, it showed 0

emericg, 2022-01-18:
Hi,
I don't have a Grow Care Garden, so the support is only theoretical. If you are willing to help test a few things, we can probably improve the situation.
Also can you provide a new screenshot, and an update of the situation?

garudaonekh, 2022-01-19:
Sure, please guide me how to test

emericg, 2022-03-10:
Sorry to get back to you this late.
If you build (in debug mode) the latest version from here and update a flower care (or flower garden ^^) you should have a trace like this one :
```
* DeviceFlowerCare update: "C4:7C:8D:62:D4:BE"
- m_firmware: "2.7.0"
- m_battery: 100
- m_soil_moisture: 8
- m_soil_conductivity: 1
- m_temperature: 17
- m_luminosity: 5810
```

You can also go to the DeviceFlowerCare::bleReadDone() function and uncomment the first two qDebug() lines, that should print the raw data sent by the sensor, like that :
```
DeviceFlowerCare::bleReadDone( "C4:7C:8D:62:D4:BE" ) on ""  / uuid "{00001a01-0000-1000-8000-00805f9b34fb}" 16
DATA: 0x "ae000086180000080100023c00fb349b"
```

That should already give me more infos about what's going on.

garudaonekh, 2022-03-12:
Ok, I will try that next week

h3rm, 2022-03-27:
@emericg sorry for offtopic here but how can i build the new versions by myself from the source code files on windows (sorry for the noob question)? :)

garudaonekh, 2022-03-28:
@h3rm why don't you just create a new issue? This one is not about compiling issue.

emericg, 2022-03-28:
@h3rm Yes you can create a new discussion for instance

Otherwise, with Qt 6.3 and up you actually need Visual Studio on Windows, otherwise the Bluetooth won't work. I think both 2019 and 2022 are fine (community edition is good too).
Because WatchFlower has no external dependencies, the process of building it is straightforward.
Install MSVC, install Qt 6.3 with the Qt installer (don't forget Bluetooth and other modules).
Load the WatchFlower.pro file with Qt Creator (MSVC and Qt 6 should be autodetected) then build :)

h3rm, 2022-03-28:
> @h3rm Yes you can create a new discussion for instance
>
> Otherwise, with Qt 6.3 and up you actually need Visual Studio on Windows, otherwise the Bluetooth won't work. I think both 2019 and 2022 are [...]

sorry, i thought this isn't really an issue and i didn't wanted to spam the issue-section with such a noob question! 🙄 sorry for that, i hope my 2nd donation yesterday let you forget that soon ^^

yesterday i tried around with windows 11 a lot of things (QT/Git/VS community/VS code/Python) but in VCS community i couldn't solve a compiler problem in the webassembly-settings because i wasn't able to install https://emscripten.org/docs/getting_started/downloads.html#installation-instructions-using-the-emsdk-recommended) with python. Also the other required tools (https://emscripten.org/docs/building_from_source/toolchain_what_is_needed.html#toolchain-what-you-need) are strange to install and use in Windows...  Maybe i will startover again with linux soon :)

emericg, 2022-03-28:
(Please create a discussion anyway so we can continue this conversation ^^)

I mean you really don't need neither python nor webassembly... And you certainly don't need visual studio code.
I don't know much about VS Code but as far as I am concerned it's a text editor, it has nothing to do with compiling WatchFlower.

If you are not really proficient with any of the software involved, you really should stick to Qt Creator.

Again, you only need :
- MSVC 2019 (community edition is ok). That's the real visual studio, not the text editor.
- Qt 6.3 with the modules, installed with the Qt Maintenance Tool, very convenient.
- Qt Creator, also installed by default with the maintenance tool.

Then open WatchFlower.pro with Qt Creator and press the play button...

emericg, 2023-03-15:
Closing, this issue is solved in v5.0

closed 2023-03-15 by emericg