emericg/WatchFlower #59: Not able to pair with the sensor

sevitz, 2022-02-22:
**Describe the bug**
I click "launch detection" but nothing happens

**Expected behavior**
I would expect it to find the sensor 

**Screenshots**
[image]

**Your environment**
Please describe the environment you are using:
1. Mac 12.2.1
2. Downloaded off your website
3. version 3.1

**Additional context**
The sensor is paired with my phone. I'm not sure if this is blocking it finding it? I've turned Bluetooth on my phone but it still doesn't find it. I've also pulled out the battery and put it back in, but that hasn't helped either.

emericg, 2022-02-23:
Yes the macOS 12 version has some real problems at the moment.
Did you try the "special" version or the regular one? You can download it from the release page or with this link:
https://github.com/emericg/WatchFlower/releases/download/v3.1/WatchFlower-3.1-macOS12.zip

sevitz, 2022-02-23:
I downloaded the version from here: https://emeric.io/WatchFlower/ 

Which looks like the same file.

sevitz, 2022-02-23:
I does connect to the Flower care app on my phone. but even if I shut that down, the device doesn't get found (and the LED doesn't blink)

emericg, 2022-02-24:
By the way do you have one of the new M1 mac? It's been reported that the app wasn't working on them...

emericg, 2022-02-24:
> I downloaded the version from here: https://emeric.io/WatchFlower/
> Which looks like the same file.

Also it's not, I just checked. The version for macOS 12 was uploaded after the initial release.

sevitz, 2022-02-25:
I do have a M1 Mac. But I've now tested with an intel Mac too (with the file from above) and same problem.

emericg, 2022-02-26:
Ok thanks for investigating this issue.
I should have a beta in a couple of days (hopefully) with compatibility work regarding new macOS and new macs, I'll update this issue when it's ready for testing.

sevitz, 2022-02-28:
👍🏼

emericg, 2022-03-10:
Can you try the new build from here : https://github.com/emericg/WatchFlower/actions/runs/1958745924
It's very much in development but it got many fundamental changes on Bluetooth and macOS support. Also there is a "device detection scanner" so you can check if any device appear using that feature :

[image]

sevitz, 2022-03-14:
[image]

Unfortunately still not seeing anything at all (And I've just synced to my phone, so it's still powered)

emericg, 2022-06-20:
Can you try with these new builds? They are mac M1 ready :)
https://github.com/emericg/WatchFlower/actions/runs/2521282029

emericg, 2022-07-22:
You can try again with the [latest builds](https://github.com/emericg/WatchFlower/actions/runs/2616589731) or the new releases, that have many improvements related to this issue.

I'm closing this thread. Feel free to reopen if the problem still occurs.

closed 2022-07-22 by emericg