emericg/WatchFlower #60: [Feature Request] Please add Xiaomi Qingping Bluetooth Smart Alarm Clock support

EcherArt, 2022-03-16:
**Is your feature request related to a problem? Please describe.**
While this device is indeed an alarm clock, it also has temperature and humidity sensor. Unfortunately I cannot use it with your app right now, but I saw Xiaomi Digital Hygrometer Clock and QingPing Temp RH Lite clock supported, so I hope it's possible for my my device to be integrated.

**Describe the solution you'd like**
WatchFlower discovering the device via Bluetooth and connecting to it.

**Describe alternatives you've considered**
Buying another device 

**Additional context**
Clock in question (pic from aliexpress) [https://ae01.alicdn.com/kf/H335313b8c5a3472c98a790ef9d09d550S/Xiaomi-Qingping-Bluetooth-Alarm-Clock-3-in-1-Temperature-Humidity-Sensor-Monitor-Mijia-Night-Light-Mihome.jpg ](url)

Clock official page from Quingping: [https://www.qingping.co/bluetooth-alarm-clock/overview](url)

emericg, 2022-03-25:
Yes I will try to add this device, but because I don't have it myself I hope that I can find the relevant information on the Internet.

EcherArt, 2022-03-29:
Thank you! I have the device so if I can help you by sending the logs/photos, I'll be good to do so.

emericg, 2022-05-12:
Ok so I think I've got this mostly implemented, I just need one more bit of infos.
Can you run _nRF Connect_ software, and get me the Bluetooth name of the device?

And if you want to be thorough, connect to the device and get a screenshot of the services, like so:
[image]

Thanks

EcherArt, 2022-05-13:
Hello, thank you so much!
Here is what I found by connecting to the device:

[image]
[image]
[image]

EcherArt, 2022-05-24:
That's awesome, thank you! 

Hopefully it'll help:

[image]
[image]

emericg, 2022-06-06:
Send me an email with your Google account (or iCloud account) and I'll add you to the testers for the new version!

emericg, 2022-06-09:
Sorry I just assume everyone is on mobile...
It's even easier to test on linux, just pick up a build from here: https://github.com/emericg/WatchFlower/actions/runs/2469137999

emericg, 2022-06-28:
Did you have a chance to test this?

EcherArt, 2022-06-28:
Yes, sorry for the late request, it does work!
[image]

emericg, 2022-06-28:
Great thanks for testing this!

closed 2022-06-28 by emericg