emericg/WatchFlower #63: Keeps crashing on Mac OS 12.3.1

White1972, 2022-05-11:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               WatchFlower [30635]
Path:                  /Applications/WatchFlower.app/Contents/MacOS/WatchFlower
Identifier:            com.emeric.WatchFlower
Version:               3.1 (03010001)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-05-11 18:58:26.3874 +0200
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Bridge OS Version:     6.4 (19P4243)
Anonymous UUID:        4A833609-E329-4A68-8332-F715FFFD4C84

Sleep/Wake UUID:       673A19AA-27B7-4D59-AF6C-5024EB9E4513

Time Awake Since Boot: 470000 seconds

System Integrity Protection: enabled

Crashed Thread:        3  Dispatch queue: com.apple.bluetooth.iobluetooth.coordinatorQueue

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [30635]

VM Region Info: 0 is not in any region.  Bytes before following region: 4389900288
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105a89000-105bbd000    [ 1232K] r-x/r-x SM=COW  ...S/WatchFlower

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff815bfb97a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff815bfbce8 mach_msg + 56
2   AppKit                        	    0x7ff818c424d8 NSWaitUntilHostTime + 102
3   AppKit                        	    0x7ff819162749 -[NSCGLSurface synchronize] + 112
4   AppKit                        	    0x7ff819161cc5 NSCGLSurfaceFlush + 83
5   GLEngine                      	    0x7ffa2df19472 glSwap_Exec + 210
6   OpenGL                        	    0x7ffa2dce29b3 CGLFlushDrawable + 59
7   AppKit                        	    0x7ff818a6d227 -[NSOpenGLContext flushBuffer] + 20
8   libqcocoa.dylib               	       0x1079749c9 0x10791d000 + 358857
9   QtGui                         	       0x107de5d90 QOpenGLContext::swapBuffers(QSurface*) + 304
10  QtQuick                       	       0x10629e497 0x1061d6000 + 820375
11  QtQuick                       	       0x10631cc87 QQuickWindow::event(QEvent*) + 839
12  QtWidgets                     	       0x106c022f6 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
13  QtWidgets                     	       0x106c036ad QApplication::notify(QObject*, QEvent*) + 477
14  QtCore                        	       0x108b43d66 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
15  QtGui                         	       0x107d91355 QPlatformWindow::deliverUpdateRequest() + 69
16  libqcocoa.dylib               	       0x10793f9f2 0x10791d000 + 141810
17  libqcocoa.dylib               	       0x10792c0cc 0x10791d000 + 61644
18  libdispatch.dylib             	    0x7ff815a7d317 _dispatch_client_callout + 8
19  libdispatch.dylib             	    0x7ff815a7fd7c _dispatch_continuation_pop + 453
20  libdispatch.dylib             	    0x7ff815a91208 _dispatch_source_invoke + 2179
21  libdispatch.dylib             	    0x7ff815a89bb1 _dispatch_main_queue_drain + 744
22  libdispatch.dylib             	    0x7ff815a898bb _dispatch_main_queue_callback_4CF + 31
23  CoreFoundation                	    0x7ff815d3c9c7 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
24  CoreFoundation                	    0x7ff815cfd93f __CFRunLoopRun + 2771
25  CoreFoundation                	    0x7ff815cfc7ac CFRunLoopRunSpecific + 562
26  HIToolbox                     	    0x7ff81e983ce6 RunCurrentEventLoopInMode + 292
27  HIToolbox                     	    0x7ff81e983a4a ReceiveNextEventCommon + 594
28  HIToolbox                     	    0x7ff81e9837e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
[... 492 lines cut ...]
  ],
  "experiments" : [

  ]
}
}

Model: MacBookAir9,1, BootROM 1731.100.130.0.0 (iBridge: 19.16.14243.0.0,0), 4 processors, Quad-Core Intel Core i7, 1.2 GHz, 16 GB, SMC 
Graphics: Intel Iris Plus Graphics, Intel Iris Plus Graphics, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x870), wl0: Jul 16 2021 18:25:13 version 16.20.328.0.3.6.105 FWID 01-30be2b3a
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 86.0

emericg, 2022-06-07:
Can you try the 4.0 beta version that can be found on the CI builders and let me know if it still crashes?

https://github.com/emericg/WatchFlower/actions/runs/2456127012

emericg, 2022-06-24:
WatchFlower 4.0 has been released, please re-open if the issue still occur.

closed 2022-06-24 by emericg