emericg/WatchFlower #64: IFTTT integration

rngarcia, 2022-05-13:
Would be great if the app could trigger IFTTT actions. This could be useful for triggering alerts on multiple devices (i.ex low moisture alert would send an e-mail) and also enable integration with smart home devices (scene triggers)

emericg, 2022-06-07:
Yes IFTTT integration would be nice.
That being said I know nothing about IFTTT (except it cost almost as much per year as the application makes :) so help would be appreciated.

WatchFlower is already more capable than it used to be. There is a version capable of using a MySQL database to be synced across multiple devices, and an MQTT version. So adding other bridges would be interesting.

emericg, 2026-04-15:
Ok let's face it, that is not planned anytime this decade, sorry. Not sure if IFTTT is even a thing anymore.