emericg/WatchFlower #67: Date choose is not possible on journal entry

andrelsr, 2022-06-26:
Hi,
When I click on the date to edit it when adding a new entry on journal, the days appear all crumpled on upper lef side of calendar. So I can't choose the date.
[image]

I didn't notice the journal before, it will be very useful.
Android, app v4.0

emericg, 2022-06-26:
Yes the version 4.0 has been released yesterday, it has many other new features than the journal, like background updates and notifications ^^

> When I click on the date to edit it when adding a new entry on journal, the days appear all crumpled on upper lef side of calendar.

Indeed it looks like this bug slip into the final release. I'll fix it tomorrow and release a new version very soon, thanks for reporting it!

emericg, 2022-06-29:
Ok version 4.1 has been released with this bug fixed.

closed 2022-06-29 by emericg