emericg/WatchFlower #68: Bluetooth disabled notification after allowing bluetooth acces M1 macOS 12.4

lesleyblankendal, 2022-06-29:
**Describe the bug**
Bluetooth disabled notification after allowing bluetooth acces macOS 12.4. When the app launches I give it permission but all devices show as offline with a notification that my bluetooth is disabled. Also granted additional access in privacy & settings > bluetooth, to see if that makes any difference.

iPad version that can be run from Mac App Store on M1 iMac doesn't have this issue

**Expected behavior**
Sensors show as online

**To Reproduce**
Steps to reproduce the behavior:
1. Open app on macOS
2. grant bluetooth access
3. Go to Home Screen
4. See error

**Screenshots**
[image]

**Your environment**
Please describe the environment you are using:
1. macOS 12.4, iMac 24 inch late 2020
2. downloaded 4.1 update just now to see if it was fixed. Bug was introduced in 4.0
3. 4.1

steveatinfincia, 2022-08-02:
I'm seeing this one too, though it works if the binary is launched from a terminal rather than through finder which is certainly strange.

jeeftor, 2022-12-14:
I also see it on an M1

jeeftor, 2022-12-14:
>This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

I got this error when I ran it from CLI

jeeftor, 2022-12-14:
Sounds like a `.plist` issue...

jeeftor, 2022-12-14:
[image]

But it looks like you have the plist stuff...

emericg, 2022-12-14:
Indeed. Can you try with a more recent build?
https://github.com/emericg/WatchFlower/actions/runs/3660319742

It has both improvements and newer Qt version, maybe that will help.

fabiang, 2023-04-16:
Still having this issue with version 5 on Mac OS 13.2.1. Also tried [recent build](https://github.com/emericg/WatchFlower/actions/runs/4637893660) with no luck.

emericg, 2023-04-20:
Yes I'm really sorry, this problem has proven difficult to understand and to solve...
But I haven't forgotten it. It's just that I failed every time I had a look at it (last time being tonight :)

fabiang, 2023-04-20:
I've also spent an hour trying to find the error. Imho this must be an upstream bug in Qt and/or OS X?

fabiang, 2023-12-03:
I think I figured it out. On my local install, I've just added this to my Info.plist:

```
<key>UIBackgroundModes</key>
<array>
   <string>bluetooth-central</string>
</array>
```

Someone able to confirm this? Have it from here: https://stackoverflow.com/a/48390949 Possible value are here: https://developer.apple.com/documentation/bundleresources/information_property_list/uibackgroundmodes

emericg, 2023-12-04:
That's interesting, but it is said that UIBackgroundModes is only for iOS, and that using it prevents you from being accepted on the app store. Weird option...

I'll test this on a mac but yes we'll need people to confirm this has an actual impact beyond placebo.

fabiang, 2023-12-07:
@emericg I've just checked if this automatically prevents you from being accepted to the Appstore and that should only be the case if your App doesn't use Bluetooth (or an Apple employee doesn't understand that is does). And from what I've read this values is definitely needed for Bluetooth to function. I've now working the fifth day with this option.

emericg, 2023-12-16:
It made zero difference in my testing, and the doc still says for iOS only.
But truth be told I don't really care enough to go against this, so I've added this line to the plist, we'll see if anyone report an actual difference in behavior...

fabiang, 2023-12-23:
Tested once again with [build 7233199927](https://github.com/emericg/WatchFlower/actions/runs/7233199927) and can confirm it's working.

NathanBnm, 2024-01-03:
I have the same issue with the last build (https://github.com/emericg/WatchFlower/actions/runs/7233199927)
[image]
[image]

fabiang, 2024-01-03:
I removed the permission manually and now I'm unable to get it working again. Very weird.

fabiang, 2024-12-27:
Coming back to this after a year...

I've added the following to Info.plist:

```
    <key>NSLocalNetworkUsageDescription</key>
    <string>Test</string>
```

and then adding the permission in the Mac OS settings manually for Bluetooth. Is someone able to confirm that it's working on Mac OS then? I'm currently running OS X 15.1.1 and Watchflower 5.4.

NathanBnm, 2025-02-26:
@fabiang I can confirm it's working fine after doing this.

I have Watchflower 5.4 on mac M1 with macOS Sequoia 15.3.1

fabiang, 2025-03-08:
@NathanBnm Could you try to remove the application and try again with the changes in Info.plist? It should not work then again. Still no idea who's fault it is then. I still guess it's how bluetooth is requested by the application.