emericg/WatchFlower #70: build instructions

svennd, 2022-07-15:
I'm trying to build WatchFlower for Android, but sadly I seem to fail. I installed Qt Creator (as this seems the only way to get qt6 installed on Linux), installed the bluetooth module and chart module. The build for arm64-v8a succeeded with some warnings and a java error (of no consequence I think) 

The generated apk I tried to install but upon opening the app, it just closes again. 

Are there any up-to-date build instructions available for Android devices ? How could I debug this (building the x86 and then emulation?) ? 

Also more of a question, would you be open to sell your builds somewhere other then google's platform such as eg. https://itch.io ? I'm not a big fan of giving google/my phone my creditcard info.

thanks !

svennd, 2022-07-15:
I'm sorry, seems I build for the wrong level of android. It seems to work now. Still I'd might be worth considering to release your updates on a platform such as itch.io or others ?

emericg, 2022-07-22:
Hi,
Do you know what API level it was building for by default? And what did you changed it to? (should be 31 or 32)

Otherwise, yes, to a certain extent I would be interested in using a third party platform to sell my softwares. It would be for the multiplatform aspect (buy once, use it everywhere) and to be independent from the big companies and their continuous string of unilateral decisions affecting my softwares and business.

But I don't know about such a platform, with good support for desktop and mobile, and not geared toward games (otherwise itch.io or humble bundle would probably be good contenders).
Also, it's impossible for iOS and damn complicated for Android...

closed 2022-07-22 by emericg

svennd, 2022-07-22:
Hey Emericg,

I tried it on my non-main phone and that was a bit older, so I set it to 29; At first I took the debug version (that was generated), but then I found an option in Qt Creator to "add" release build. (in CMake file?) (not the build release button mind you) --> anyway a clear lack of knowledge on my part. 

Yeah, this is clearly allot of work & dedication, and its a bit sad you only get to have revenue from store's. (that take 30%) I flat out refuse to buy from android/ios store, but I understand that probably aint the majority of users ...  thanks for considering!