emericg/WatchFlower #73: Impossible to export data - Android doesn't find any app.

Beardmix, 2022-09-19:
Thanks a lot for this amazing app. It works generally super well and has a lot of great features!
I downloaded it mainly to be able to export data, but the functionality doesn't work on my Android phone :/ 

When I go to settings and click on "Export File" at the bottom, the Android share menu opens with a messages saying that "there are no apps that can handle this action". I think that it is because of the line:
`Settings.qml#L1672`
`utilsShare.sendFile(file, "Export file", "text/csv", 0)`
I think the problem is "text/csv".
I tried downloading many CSV apps and File managers but nothing works. 

I feel that this could be the answer:
https://stackoverflow.com/a/72105124/5539469

If not you could also save the file in the non-protected area of the phone so that a file manager could retrieve it no matter what.

I hope my description is accurate enough and I am willing to test some fixes if you need a tester :D 

Thanks again and have a nice day!

emericg, 2022-11-24:
Hi, sorry I have not been working on WatchFlower for a while, but I'm planing a release soon.

That's weird, but I guess that could happen. The mime type is already set to text/csv, so I guess the problem is somewhere else. Can I ask what version of Android you are using?
On my phone, almost every app installed offer to open this file :/

There used to be an option to save a file on the memory card, but this is a feature that got removed from Android, and ban from the store.
I will try to implement a workaround for the next release though.

Beardmix, 2022-11-25:
Hi Emeric,

No worries for the time.

I still think that the mime type could be problematic.
From the link I sent above:
> intent.setType("text/csv"); does not work on regular file explorers, i can pick any file from them. But it does have effect on Dropbox app, which only shows .csv files. How to do same for regular file explorers?
>
> Try intent.setType("text/comma-separated-values")

I think that "text/*" or potentially "text/comma-separated-values" would work better, at least I could save the file in Drive or something. But of course, it would filter fewer apps.

I am running Android 13 and running WatchFlower 4.1.

Unrelated:
In the meantime I created my own version of the app but I weirdly get disconnected after a few seconds by the device while I am synchronizing. This problem doesn't appear to happen in your app but I couldn't see what was different in your code.
Have you encountered this problem? If so how did you solve it?

emericg, 2022-11-25:
> I am running Android 13 and running WatchFlower 4.1.

That is most likely the problem, it does require new things related to file handling.

> Have you encountered this problem? If so how did you solve it?

Some BLE devices like the Flower Cares times out after a few seconds by design.
To stay connected for longer period of time, you need to perform some kind of handshake with the device. While it's not properly documented, you can check my code in the file https://github.com/emericg/WatchFlower/blob/master/src/devices/device_flowercare.cpp

Beardmix, 2022-11-28:
Make sense.
Thanks for the feedback and pointer to the handshake.

Beardmix, 2022-11-28:
Sorry to continue a bit out of thread, but seeing your implementation of the handshake it seems that it works only on Android, as the BLE MAC address is necessary. 
How can it work with the ID that iOS returns instead of the device MAC?

emericg, 2022-11-29:
> Sorry to continue a bit out of thread

No problem. You really need the MAC, the UUID that iOS gives you is completely random.
Thankfully the FlowerCare broadcast their MAC in their service data, as you can see in this doc https://github.com/emericg/WatchFlower/blob/master/docs/mibeacon-ble-api.md

emericg, 2023-03-27:
Can you maybe try the latest version on the play store (the 5.0) and check if anything changed regarding your issue?

Beardmix, 2023-03-28:
It works like a charm 👍
Good work!

emericg, 2023-03-28:
Alright that's great news!
Closing this issue.

closed 2023-03-28 by emericg