emericg/WatchFlower #74: Parrot Flower Power no data

giosal, 2023-01-15:
**Describe the bug**
Managed to connect to Parrot Flower Power. It's syncing, when clicking flash LED, it's flashing. However, there is no data from sensors.

**Expected behavior**
Should display the data from sensors, for instance, temperature should be +5 C right now

**To Reproduce**
1. Turn on Flower Power.
2. Launch Watch Flower app.
3. Go to Home screen and wait for the app to see Flower Power
4. Click on Flower Power and click Sync

**Screenshots**
[image]

**Your environment**
Please describe the environment you are using:
1. Windows 11 x64 build 22623.1095 Insider Preview Beta
2. Github
3. 4.1

emericg, 2023-01-24:
Indeed that's not good.
I'm going to try to speed up the connection between the app and the Flower Power device, we'll see if that's why it's not working for you.

In the meantime, do you think it would be possible for you to scan your sensor with this application (https://github.com/emericg/toolBLEx/actions/runs/3998486641), export the results and send it to me? I know they are multiple firmware for the Flower Powers that aren't communicating their results in the same way, but I only have one to test (it's stuck on version 1.0.0)

giosal, 2023-01-25:
Thank you for your reply.
I was able to connect and save the Flower Power in the device list and have been since able to track the data.
However, there are several issues still:
- Connection and syncing takes more than 30 sec with a distance of less then 2 meters between computer and the device
- the "Low battery" icon is shown at any battery level, including full battery
- Luminance data always has the same value (130 lux)
I'm on firmware 1.1.0. If you find a way to update the firmware, I do have .bin file for v2.5.0
[image]
As for toolBLEx output, please find it attached
[Flower_power_2FCF.txt](https://github.com/emericg/WatchFlower/files/10503558/Flower_power_2FCF.txt)

Narga93, 2023-02-17:
Hey! My god, sorry for the long delay. Tomorrow Ill send the data u asked.
Really sorry, lot of changes in work and education.

Hope u well,

João de Almeida Barata

emericg, 2023-04-06:
> Connection and syncing takes more than 30 sec with a distance of less then 2 meters between computer and the device

Yeah these sensors just don't have good Bluetooth at all. From what I gather, the official application doesn't work any better than WatchFlower, probably way worse...

> the "Low battery" icon is shown at any battery level, including full battery

That is weird, maybe the firmware 1.1 has a bug? the Bluetooth battery service does have a value of 0 in the device scan you uploaded, that would certainly explain the red icon.

> Luminance data always has the same value (130 lux)

They also have bad luminance sensors, but maybe my computation are wrong :)
Firmware version 2 IIRC have "direct" luminance value instead of relying on a computation, but I am also stuck on 1.0.0.
I have a Parrot Pot with a recent firmware, it also has terrible luminance sensing.

> I'm on firmware 1.1.0. If you find a way to update the firmware, I do have .bin file for v2.5.0

Sure, you can upload it here, but I don't know if I'll ever be able to flash that.

> As for toolBLEx output, please find it attached [Flower_power_2FCF.txt](https://github.com/emericg/WatchFlower/files/10503558/Flower_power_2FCF.txt)

Thank you for taking the time to upload this.

emericg, 2023-04-06:
> Hey! My god, sorry for the long delay. Tomorrow Ill send the data u asked.
Really sorry, lot of changes in work and education.

You are probably responding to the wrong thread ^^ No worries though, if you ever find some time let me know.

emericg, 2023-04-20:
Closing this, as the initial issue seems resolved anyway.

closed 2023-04-20 by emericg