emericg/WatchFlower #80: [Feature Request]

howels, 2023-08-08:
Please add battery status for the Xiaomi flower care devices. These have monitoring of the internal CR2032 battery status.

Would be very useful to know when to replace batteries.

emericg, 2023-09-28:
By the way, sorry I never responded but I didn't understand your request.
Battery levels have always been in the app, in the main screen and in the sensor infos?!

[image]

[image]

howels, 2023-09-28:
> By the way, sorry I never responded but I didn't understand your request. Battery levels have always been in the app, in the main screen and in the sensor infos?!
> 
> [image]
> 
> [image]

I do see the battery level on one device but only one out of 4 identical devices shows the battery level.
Also, on Android the app somehow overlaps with my status bar which nothing else does.
[image]
[image]
[image]

emericg, 2023-09-29:
> I do see the battery level on one device but only one out of 4 identical devices shows the battery level.

I'm not sure what to say about that, the battery level won't show if the value from the flower care is really weird (not in the 0-100% range), so it's possible the problem is on the sensors.

> Also, on Android the app somehow overlaps with my status bar which nothing else does.

Damn I'm really sorry about that, it's been a really tricky issue on the last couple release. I guess your phone has no notch?
I'll be doing a release in the next couple days.

howels, 2023-09-29:
> I'm not sure what to say about that, the battery level won't show if the value from the flower care is really weird (not in the 0-100% range), so it's possible the problem is on the sensors.
>
>
> Damn [...]

Hmm. I used to get erroneous low battery readings just after replacing the CR2032 cells, possible that the sensor is incorrectly calibrated.

I run a Sony Xperia 1 III, no cutout here.

emericg, 2023-10-03:
> Hmm. I used to get erroneous low battery readings just after replacing the CR2032 cells, possible that the sensor is incorrectly calibrated.

Yeah that would certainly make sense, either the FlowerCares are not reading the battery voltages correctly, or the CR2032 cells you are using are not working in the usual range (but that would be weird though).

> I run a Sony Xperia 1 III, no cutout here.

Ok I've pushed a new version to Android beta channel, it will be published as soon as it passes the store automated checks, can you take a look at it and see if it fixes your issue?

closed 2023-11-22 by emericg

howels, 2023-11-22:
Apologies for not replying - I see a battery warning now on one device. I think my units have inaccurate voltage sensors or something.

emericg, 2023-12-04:
No problem, the battery problem probably wasn't like we discussed, and the header problem has been solved by the 5.3 release a while ago, that's why I closed this issue.