emericg/WatchFlower #82: German Translation

Roughtrade, 2024-10-16:
[image]

emericg, 2024-10-16:
So replace relative humidity (different from soil moisture, bodenfeuchte) with luetfeuchte?
There already are two different strings for the two terms, but maybe that was translated by someone who didn't have an hygrometer.

Roughtrade, 2024-10-16:
Not "luetfeuchte"  rather "Rel. Luftfeuchte" 4 the qingpin.
Mit freundlichen Grüßen / best regards
________________________________________
Peter Rene´ Beauvain
Leisniger Str. 15
D-12627 Berlin
fon   :    +4930 265 40743
mobile: +49179 4444 588
fax   :    +4930 223 411 791
eMail : ***@***.***

emericg, 2024-10-17:
Alright but that's a bit verbose though, is it how hygrometers (apps or otherwise) usually label their relative humidity in German?